wled/WLED #2670: Legacy browser support for ones not supporting conic-gradient

hockeymikey, 2022-05-14:
### What happened?

The color wheel will not show for browsers not supporting conic-gradient such as Waterfox Classic. I assume https://github.com/Aircoookie/WLED/commit/3dec4a6651774ab5e62103dbec138136513e16eb#diff-3fc7dff92571265d985e6ee7e1daf723e8b0ac4d2c5929837f2c52e28e60706f= is when it was introduced, but that iro.js is an ugly one liner so not positive.

### To Reproduce Bug

Use Waterfox Classic or other older browsers

### Expected Behavior

You see the color wheel

### Install Method

Binary from WLED.me

### What version of WLED?

0.13.1

### Which microcontroller/board are you seeing the problem on?

ESP8266, ESP32

### Anything else?

Not sure how you were doing it before, because again that one liner, but I think a good compromise would be to add some browser compatibility as simple as just checking the user agent version and using either the conic-gradient or something else for the background.  

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2022-05-15:
Not a bug.
You will be able to switch to a hue slider instead of the color wheel in an upcoming release.

hockeymikey, 2022-05-15:
@blazoncek Yes it is a bug, it broke older browsers with an update. I'm not looking for an enhancement, just a restoration of how it should work and it does in new browsers and how it used to. And how does a hue slider help or is relevant to using CSS not supported in older browsers?!

blazoncek, 2022-05-15:
Things move forward and require changes.
You may consider it a bug if you wish, but we decide what is a bug and what isn't one.

Still, if you want old browser support just use old version of WLED.

hockeymikey, 2022-05-15:
@blazoncek The old implementation did not require change. And again if one desires to use the new css, that's fine but there is no reason not to retain the older functionality unless reading an user agent is too difficult which I really do not think it is. Can't hand wave this away with "progress", I think the reason for the change was simplicity vs the older function but that's really it. Also how does one even edit that minimized one liner. Why is it not formatted properly? 

Doesn't matter what you consider a bug, this is factually a bug. You introduced a change that broke functionality. That's a bug, full stop. An "enhancement" is not fixing stuff that is broken, specially this is a defect and/or regression, both of which are types of bugs. Now one could say a sub-issue adding support for reading user agents could be an enhancement, no argument there if it is spun off, but if it's coupled with changes fixing a regression in code, such as this, then it's not an enhancement, it is a bug because bugs fix broken things, and the larger scope navigates away from just adding to a larger narrative of the broken functionality. 

It's disingenuous telling users to use an older version just to get it to work in browsers. And these aren't old browsers, lets make that a fact, Browsers like Waterfox or say Palemoon are very popular (relatively) and not old. Further, the solution is already there since again this is a regression. Literally the only thing that needs to be done is adding a conditional for certain browser user agents to determine if you apply CSS or if you build up a svg using paths in the g tag.

Aircoookie, 2022-05-16:
We test for an aim to support the latest versions of Firefox, Safari, and Chromium-based browsers.
As such, I cannot guarantee functionality with Waterfox unfortunately, especially in regard to newer JavaScript features (ES6,...)

The reason why the color wheel code is minimized is that it is an external library, [iro.js](https://iro.js.org/). Please consider raising an issue with them, and once they fix it, we could update to that.

blazoncek, 2022-05-16:
> Literally the only thing that needs to be done is adding a conditional for certain browser user agents to determine if you apply CSS or if you build up a svg using paths in the g tag.

Then why don't you do it instead of complaining? After all WLED is an **Open Source Software** and perhaps a few will benefit from your work. If you share it as we do, that is.
And you can *thumbs down* as much as you like. It is not changing the fact that >99.9% of users are comfortable with current implementation.

hockeymikey, 2022-05-16:
@blazoncek Knew that line was coming, lmao. 
@Aircookie Thank you! That was very helpful and makes sense now. I will do just that.

hockeymikey, 2022-05-16:
How about adding https://projects.verou.me/conic-gradient? Adds support for legacy browsers.

Issue in question on their end: https://github.com/jaames/iro.js/issues/186

softhack007, 2023-10-07:
Unfortunately we have limited ressources, and many things on our list are much urgent. Honestly, I don't see how we can add (and maintain) support for additional browsers in this situation.

Closing as won't fix.

closed 2023-10-07 by softhack007