wled/WLED #2672: Multi Relay Usermod pin config reset after reboot

1MisterT, 2022-05-18:
### What happened?

I compiled a custom version of WLED 0.13.1 via VS Studio (for  esp32s via esp32dev and esp8266 nodemcu via d1_mini) and added the MultiRelay usermod with 2 Relays.
And once I open the usermod settings in WLED GUI and set some value for the relay pin (e.g. **5**), save it and reopen it, the pin settings are reset and back to **-1**.

When I try to control the relay via [JSON API ](https://kno.wled.ge/features/relay-control/) or HTTP, it works as long as I do not restart the ESP. Once restarted the HTTP API says `"NoOfRelays":2,"0":-1,"1":-1,"error":"No active relays","SW Version":"0.1.3"}` 

### To Reproduce Bug

- Compile WLED 0.13.1 from VS Studio
- add Usermod MultiRelay like explained [here](https://github.com/Aircoookie/WLED/tree/main/usermods/multi_relay)
- choose `default_envs = esp32dev` as default env
- upload, go to usermod and change relay 0 pin to 5
- save and reopen usermod settings pin should be -1 again

### Expected Behavior

I expected the pin setting to stay 5/ the relay to still bein defined after restarting the esp

### Install Method

Self-Compiled

### What version of WLED?

0.13.1

### Which microcontroller/board are you seeing the problem on?

ESP8266, ESP32

### Relevant log/trace output

```shell
FileRead: /cfg.json
MultiRelay:(-1):enabled = false
MultiRelay:(-1):enabled = true
MultiRelay:(-1):broadcast-sec = number
MultiRelay:(-1):broadcast-sec = 0
MultiRelay:(7):pin = int
MultiRelay:(7):pin = 5
MultiRelay:(7):active-high = false
MultiRelay:(7):active-high = true
MultiRelay:(7):delay-s = number
MultiRelay:(7):delay-s = 0
MultiRelay:(7):external = false
MultiRelay:(7):button = number
MultiRelay:(7):button = -1
MultiRelay:(7):pin = int
MultiRelay:(7):pin = 4
MultiRelay:(7):active-high = false
MultiRelay:(7):active-high = true
MultiRelay:(7):delay-s = number
MultiRelay:(7):delay-s = 0
MultiRelay:(7):external = false
MultiRelay:(7):button = number
MultiRelay:(7):button = -1
MultiRelay:(-1):HA-autodiscovery = false
MultiRelay:(-1):HA-autodiscovery = true
MultiRelayPIN ALLOC: Pin 5 allocated by 0x0d (13)
PIN ALLOC: Pin 4 allocated by 0x0d (13)
 config (re)loaded.
MultiRelayPIN ALLOC: Pin 5 already allocated by 0x0d (13)
PIN ALLOC: Pin 4 already allocated by 0x0d (13)
 config (re)loaded.
JSON buffer released. (5)
Writing settings to /wsec.json...
JSON buffer released. (4)
Writing settings to /cfg.json...
MultiRelay config saved.
MultiRelay config saved.
JSON buffer released. (2)

debug log from esp32 after saving usermod config in WEB GUI
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2022-05-18:
Something is messing up  your pin allocations.
Check that.

I am using it without issues.

blazoncek, 2022-05-18:
Check pin manager.

1MisterT, 2022-05-21:
0x0D is WLED_Multirelay, so its basically already allocated by mutlirelay

blazoncek, 2022-05-21:
Try other pins. 4 & 5 are usually I2C and may come preallocated.

closed 2022-11-01 by blazoncek