wled/WLED #2676: WWCWA LED strip support

bergdahl, 2022-05-22:
**Is your feature request related to a problem? Please describe.**
I recently bought a Warm white/Cold white/Amber LED SK6812 strip for ambient lighting and got frustrated when trying to get it hooked up with Home Assistant. There were no support in any of the standard firmwares supported by HA, including WLED.

**Describe the solution you'd like**
Support for WWCWA strips in WLED.

**Describe alternatives you've considered**
I have found no projects that has this functionality. Animating the three different "white" colors does create really nice evening effects, as long as the firmware understands what it is doing color temperature-wise.

Thank you for your ideas for making WLED better!

blazoncek, 2022-05-22:
You can use them as regular (RGB) addressable LEDs.
Then instead of Color wheel (or with it) enable RGB sliders in UI.
You will be able to control individual white LED as a separate channel (R or G or B).

bergdahl, 2022-05-22:
I am aware I can use it that way, but using color sliders or a color wheel to set color temperature does not make much sense.

blazoncek, 2023-05-27:
Closing as a duplicate of #552

closed 2023-05-27 by blazoncek