wled/WLED #2680: Add feature of sync!

ChristianSwift, 2022-05-27:
Espressif had realease a protocal called [esp now](https://www.espressif.com/en/products/software/esp-now/overview). it can transmit data between two or more esp chips. I hope you will add this protocol to firmware for fast and efficiency sync function!

huggy-d1, 2022-07-20:
From the 2016 dated pdf from Expressif:
```
ESP-NOW supports the following features:
• Encrypted and unencrypted unicast communication.
• Mixed encrypted and unencrypted peer devices.
• Up to 250-byte payload can be carried.
• The sending callback function that can be set to inform the application layer of
transmission success or failure.
ESP-NOW technology also has the following limitations:
• Broadcast is not supported.
• Limited encrypted peers. 10 encrypted peers at the most are supported in Station
mode; 6 at the most in SoftAP or SoftAP + Station mode. Multiple unencrypted peers
are supported, however, their total number should be less than 20, including
encrypted peers.
• Payload is limited to 250 bytes.
```
Is 250 bytes per packet enough?  Are 10 paired peers enough with NO ability to control via normal WLED means?  Are 6 paired peers enough with ability to slip in random WLED control requests between all the other traffic?  Or, if most pairs are left unencrypted, less than 20 total peers, is that enough?

Any one of them might be enough to make this a viable FORK of WLED, but combined, it seems like that's probably the best option.  Create a fork of WLED with the intent of supporting ESP NOW with a built-in playlist of some sort, perhaps written to an SD card (see @Quindor's or other ESP32 SD card modules).

blazoncek, 2023-09-19:
Implemented and will be available in 0.15.

closed 2023-09-19 by blazoncek

Rushmed, 2024-03-21:
Thanks for adding this feature.
Got 0.15.0-b1 installed on ESP8266.
But only see:
[image]
Isn't this feature available for 8266?

blazoncek, 2024-03-21:
You need to enable it 1st in wireless settings.

Rushmed, 2024-03-21:
Thx. Ill try.

eagleco, 2024-12-19:
> Thx. Ill try.

Did you get it working? All I can find in the new v15 documentation about esp-now is related to remote control, but nothing about sync.

blazoncek, 2024-12-19:
Combining ESP-NOW sync and WiFi to work seamlessly together is not trivial.
Testing so far proved to be one or the other not both together. Unfortunately.
I am still working on it but have shifted progress into my fork. There is an option available form @steveeisner which is implemented as a mix of usermod and core extensions tailored for his particular needs.

Erik-JanWind, 2025-01-10:
I have 8 devices on  batteries that are in AP mode en all connected with a wiz remote. How can I use ESP now sync in AP mode? It's checked and I checked on 1 instance that they all need to sync to one controller, but it does not work.

DedeHai, 2025-01-10:
@Erik-JanWind please use discord or discourse forum for support questions, github is to track code issues.

Erik-JanWind, 2025-01-10:
> @Erik-JanWind please use discord or discourse forum for support questions, github is to track code issues.

Allright, will do, sorry!