wled/WLED #2682: Animated Staircase usermod preset switch bug

H0nzik1, 2022-05-31:
### What happened?

I am using Animated Staircase usermod which stops working after preset is switched to another one. Only solution is to go to LED settings and press Save button ( http://192.168.50.90/settings/leds? ), then usermod starts working again. 

### To Reproduce Bug

1) enable Animated Staircase usermod
2) create multiple presets like different color and so on
3) switch preset to another one
4) Animated Staircase mod stops working. Even no response for MQTT command to swipe UP/DOWN to turn on the lights.
5) Go to "LED Preferences" menu http://192.168.50.90/settings/leds? and hit "Save" button
6) Animated Staircase usermod is finaly working again with new chosen preset.

### Expected Behavior

Expected behaviour is to skip steps 4) and 5) so after changing preset Animated Staircase usermod works instantly.

### Install Method

Self-Compiled

### What version of WLED?

WLED 0.13.1

### Which microcontroller/board are you seeing the problem on?

ESP32

### Anything else?

I have tried also latest github version, no luck too.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

H0nzik1, 2022-05-31:
Solved: Animated Staircase usermod does not work with Transition time 0s. Is it bug or feature that Animated Staircase usermod does not work with 0 transition time? There are two places to set transition time, main page (in seconds) and in LED preferences (in miliseconds) for each preset.

blazoncek, 2022-11-18:
Animated Staircase has been updated to work with 0.14. If the issue is still present, please reopen it.

closed 2022-11-18 by blazoncek