wled/WLED #2691: nonPWM White (or RGB(CCW))

roboraptor, 2022-06-18:
**Is your feature request related to a problem? Please describe.**
Sometimes i want to control a relay with just ordinary lightbulb socket or something and using "PWM WhiteL on relay input is not really a nice/working solution.

**Describe the solution you'd like**
In LED Preferences ->Add Segment -> Led Type: 1/0 White
On "Controls" page when segment is choosen -> Some kind of ON/OFF switch (instead of dimming slider)

**Describe alternatives you've considered**
I tried to use presets on the segment but it didnt really satisfied me with the ease of use.

**Additional context**
This switches can be added even to RGB or CCW. 
I know it will lack the transition and fading but sometimes it is not really the thing you actually need.

blazoncek, 2022-06-19:
You can use Multi Relay usermod.
Then create presets to control individual relay.
https://kno.wled.ge/features/relay-control/

roboraptor, 2022-06-19:
And when I don't want to use presets?
Because I can already use presets with pwm0% and pwm100%...

blazoncek, 2022-06-20:
I am a bit intrigued how you use WLED without presets to be honest. But to each its own, as they say...
If you can code, you can enhance `bus_manager.h` to implement BusOnOff since WLED already knows about such output.
If not, you can try to find someone who can and have them make a PR. There is no immediate plan to do that from our side.

roboraptor, 2022-06-22:
Ok, I need to say I'm sorry.
There is actually an easy accessible button for relay in multirelay usernod. Just inside the info section.
That's pretty neat. I just had to spend tedious hours of fighting with patformIO and try it for myself.
I actually use presets. But sometimes it is easier to just have some buttons on front page to just do one thing.

blazoncek, 2022-06-22:
BTW I have implemented BusOnOff since it was rather trivial.
It is available in source from my fork (currently in audioreactive branch) where most of development is done. @srg74 also has recent binaries in his Weomos shield repository (experimental folder).

Controlling relay from preset is describe in KB on link I provided above.

roboraptor, 2022-06-22:
Sound Reactive... That is also the place where I am trying to get.

I will try your branch, thank you :)

blazoncek, 2022-06-22:
Not SoundReactive. That is another fork, completely separate not maintained by me.

roboraptor, 2022-06-22:
And is SR like "friendly" fork at least? You know like discussing and sharing ideas? Or totally different like "not going to touch that"

Ad your implemented busonoff, is it only in the bus_manager.h?

TysTies, 2022-09-22:
could one of you send a link to that audioreactive branch? a can't find it.

blazoncek, 2022-09-22:
just grab 0_14

TysTies, 2022-09-22:
I don't know what you mean I'm new to github. can you just send me a link?

blazoncek, 2022-09-22:
Join Discord lot's of people there. Very helpful.
And you'll get binaries there.

TysTies, 2022-09-22:
I can't I don't have a phone and I'm on a school computer.

blazoncek, 2022-09-22:
https://github.com/srg74/WLED-wemos-shield/tree/master/resources

closed 2023-02-14 by Aircoookie