wled/WLED #2694: Not displaying proper amount of pixels.

jaywehner, 2022-06-20:
### What happened?

I have WLED latest running on a wemos D1. I have three ws2815b strips connected end to end. 12V strips with a buck converter for the 5v wemos D1. Running on a 3 Amp 12v powersupply.

When I set the amount to the actual number of leds 446 it turns them all on. I save the settings and when the unit power up's it still shows 446 led's but only two strips light up and one more led on the third strip. 

If I change the number of LED's to 200 then back to 446 everything works fine. Until I unplug the power. It reverts to not working again and have to do the same trick again.

I tried changing it to 445 then back to 446 and that doesn't work. It's odd that a lower number works (200). I have not tested it with different lower values. But higher values do not work (447, 500, etc..)

### To Reproduce Bug

I tested on other strips with the same result.

### Expected Behavior

I expected the lights to light up with the requires amount.

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.13.1

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

devoh747, 2022-06-21:
Where are you injecting power on the strips? (only at the beginning of the
line? or other places as well? )
Do you have a voltmeter?  If so what voltage are you seeing at the end of
the strip between gnd and +v ?   What color are you setting the leds to ?
Is it a solid or pattern?

blazoncek, 2022-06-21:
Have you checked how your segments are set up?

jaywehner, 2022-06-21:
Single segment. Set the amps to 3.0 I will try 2.5 and test. Power is at beginning and end. But I had the issue with power only one end or both ends. It happens on any brightness, color or pattern. I'll report back voltage measurements later today.

CollinNHays, 2022-07-18:
I am also running into this issue. Using an ESP 32 and 184 WS2811 LED's on GPIO 4. After a reboot only the first 100 LEDs light up. Changing the number of LEDs to a number of 100 or less and then back to 184 is the only way to fix it for me. It is clear that this is not a power issue as the effects such as wipe immediately begin again on the first LED after ending on the 100th LED. 
[image]

blazoncek, 2022-07-18:
Check what is saved in your preset.

max-xface, 2022-07-18:
I solved my problem, I made a preset (when there were 60 LEDs on the tape in total) and the preset was assigned when turned on, and in its settings it’s worth just turning on the first 60
I recreated the preset and everything began to work well, as an option, turn on the default preset and check

CollinNHays, 2022-07-18:
> I solved my problem, I made a preset (when there were 60 LEDs on the tape in total) and the preset was assigned when turned on, and in its settings it’s worth just turning on the first 60 I recreated the preset and everything began to work well, as an option, turn on the default preset and check Or check in preset this params "start": 0, "stop": 60,

Creating a new preset solved my issue as well. I did not expect this to be something stored in a present as I have never used segments.

closed 2022-11-01 by blazoncek