wled/WLED #2695: Single LED Peek not working

roboraptor, 2022-06-21:
### What happened?

If there is only one LED set, Peek function is not working.

### To Reproduce Bug

Flash your ESP
Go to LED Preferences
Set LED stril length to 1led OR use "RGB PWM"
Go to controls
Click Peek button
*Peek line stays black*

### Expected Behavior

Peek line working even with one LED set.

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.13.1 (didnt worked earlier either) 

### Which microcontroller/board are you seeing the problem on?

ESP8266, ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2022-06-22:
I am not sure peek was designed for 1 LED only...

roboraptor, 2022-06-22:
I'm not sure why. 
It could be whole line showing color...

Also... The gradient is SO HUGE when I have only 2 LEDs set :D 
[image]

tablatronix, 2024-06-15:
This should be fixed, I cant imagine it would be hard to do.

blazoncek, 2024-06-15:
> This should be fixed, I cant imagine it would be hard to do.
> 
> I might take a look at it when i have time.

By all means!

tablatronix, 2024-06-16:
I just flashed dev and this works, I think this is already resolved. Will test in app etc.