wled/WLED #2697: Hide effects that doesn't make sense for single LED or PWM RGB+CCT?

fribse, 2022-06-26:
**Is your feature request related to a problem? Please describe.**
I have some lights controlled by WLED that are PWM RGB+CCT, some of the effects doesn't make sense (eg. bouncing ball).
I think it would be cool to just hide those effects.

**Describe the solution you'd like**
When the type of RGB is set, or the number of LED's are 1, then hide the effects that only works for a digital LED strip.
A setting for hide / reveal would be a great supplement, so people that wants to see the unusable effects can still see them if they insist.

**Describe alternatives you've considered**
None

**Additional context**

blazoncek, 2022-06-27:
This may be difficult ATM since effect (function or description) does not provide relevant data.
Some work has been made in that direction but as of now there is still no distinction which effect would (not) work with 1 pixel only.

fribse, 2022-06-29:
Hi @blazoncek 
Well, I think that's what a feature request is all about :-D Something that isn't possible at the moment, but might be doable in the future 😆 
But it would be a single bit marking for an effect, and then some logic to handle it, and I think it's great that 'there has been done some work' 😉

blazoncek, 2022-06-29:
Feature request has nothing to do with *things doable in the future*. 😉 It is a request.
What might look simple to uneducated mind may require months of work. 😉 And I am not saying it might or might not get done.
Your request is valid though. 😄

DieNummerEins, 2022-11-01:
@fribse 
May I ask you something offtopic (maybe somehow via private message)?
How exactly did you wire "WLED" to PWM RGB+CCT (what about MOSFETs etc.)?

blazoncek, 2023-03-27:
0.14.0-b2 build 2303240 has improvements regarding 1 pixel segments.
If you spot incompatible effect or a missing one, please open a new issue.

closed 2023-03-27 by blazoncek