wled/WLED #2698: Adalight via serial glitches/flicker (after pixel 85)

sh00bx, 2022-06-26:
### What happened?

Using Adalight via serial, I'm getting intermittent glitches towards the end of the signal going to the ESP.
Out of my 116 pixels, the ones up to pixel 85 are fine, pixels > pos 85 occasionally flicker in random colors while the others remain accurate (can add a picture if needed). The flicker is noticable @ peek on web GUI as well.

Physical issue with installation has been ruled out.
I managed to narrow the cause down to interference by wifi activity, proportional to traffic from/to wled:
- Light glitching triggered by looking at peek preview
- Massive glitching when uploading new binary to wled for ArduinoOTA
- Disabling wifi on the ESP works around the issue

I mostly tested on ESP32, but similar problems are noticable on ESP8266, but behavior is slightly different there (other pixels affected)

Random note: 85x3colors=255, so it might be related to some overflow (?)

Some further research lead me here:
https://www.reddit.com/r/esp32/comments/inbgjn/led_ws8211_glitching_while_using_wifi_solved/

Could integration of [FastLED-idf](https://github.com/bbulkow/FastLED-idf) (or similar) be a solution?

### To Reproduce Bug

Send Adalight to WLED via serial with >85 pixels

### Expected Behavior

No flicker / accurate colors on pixels >85

### Install Method

Self-Compiled

### What version of WLED?

WLED 0.13.2-a0 (build 2203191)

### Which microcontroller/board are you seeing the problem on?

ESP8266, ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

ChuckMash, 2022-07-06:
May also be related to how the message-length bytes inside the incoming data packet are being decoded.
I ran into the (seemingly) same issue with how the these bytes were being set while sending data out from WLED over serial. (#2628).
If that is the case, the code that deals with that seems to be around [here](https://github.com/Aircoookie/WLED/blob/main/wled00/wled_serial.cpp#L135-L149)

sh00bx, 2022-07-06:
Yup I think I recall reading you mention it somewhere 🙂 And I agree meanwhile that it may not be related.. I'm not a great coder but I managed to butcher wled_serial.cpp down to a basic fasted sketch at some point to narrow this down. Issue persisted.
So we might actually be looking at an upstream issue with wifi on esp here...

berkut0, 2024-08-14:
https://github.com/dmadison/Adalight-FastLED

I have compiled this code on the controller and the behaviour is identical!

I seem to remember something like this in relation to FastLED and it had a bug in it.

closed 2025-04-04 by github-actions[bot]