wled/WLED #2699: Feature Request: Create a segment using multiple pixel sets

aaronganek, 2022-06-27:
I would like to be able to create a **segment** with a gap in the middle.

For example, if I have a 150 LED strip. Have one single continuous animation running from across 1-50 and 101-150 (but ignore 51-100).

Instead of having only an LED **start** and **end**, I would would like to be able to set multiple starts and ends to form a single segment.

This has been such a cool hobby! Thanks for your hard work that makes it possible.

blazoncek, 2022-06-27:
Use ledmap and reposition LEDs logically.
Look into [KB](https://kno.wled.ge) for examples.

closed 2022-06-27 by blazoncek

aaronganek, 2022-06-27:
Thank you @blazoncek 

I would still like to request non-sequential segments inside the WLED interface.

The LEDmapping json is definitely helpful, but I would love it if it were dynamic.

Sometimes I would like run a continuous effect on LEDs 1-100. And other times 1-50,101-151.

I love that the WLED makes it easy to design and recall different effects with presets. Would love it if I could do that with LED pixel order as well.

Thank you. Even with just your insight about remapping LEDs is already helpful. I really appreciate the advice.

aaronganek, 2022-06-27:
You rock! Perfect, wow. Thank you! @blazoncek

Related question, but unrelated to the feature request: Can I map two LEDs to one virtual pixel? (essentially, making it as if the LEDs are virtually wired in parallel)

blazoncek, 2022-06-28:
If you can use the middle part as a segment use ledmap that shifts pixels so that the ones of same color are one next to the other then use grouping of 3 in that segment.
If you need one continuous segment you can rearrange wiring of that part.