wled/WLED #2704: Raspberry Pi Pico W released..

Doyle4, 2022-06-30:
The Raspberry Pi Pico W has been released and would be amazing to see WLED come to it :)

https://thepihut.com/collections/featured-products/products/raspberry-pi-pico-w

blazoncek, 2022-07-01:
Please contact all of the authors of libraries that WLED uses to port them for RPi Pico W.
After they are ported it may be possible to port WLED as well.

Libraries include:
FastLED
NeoPixelBus
ESPAsyncWebServer
ESPAsyncTCP
ESPAsyncUDP
IRRemoteESP8266
LittleFS
and many others.

We are currently too busy adding features to current version of WLED.

softhack007, 2022-07-01:
Please add "Espressif IDF" and "arduino-esp32" to that list 😉.

It seems that the Pico W is an "ARM Cortex-M0+ cores clocked at 133MHz; 256KB RAM" -  a completely different architecture. The low RAM and slow processor might be problematic, unless we try to create "WLED light".

I suggest to close this issue, at it seems out of reach at least for the next year.

Maybe this project would be easier for a start: https://github.com/kitesurfer1404/WS2812FX

closed 2022-07-01 by Doyle4