wled/WLED #2710: Mobile Responsiveness Issues in Settings

FuriousGeorge91, 2022-07-08:
### What happened?

You are able to scroll left and right in all the settings screens except settings_leds. On desktop the current responsive settings cause the text to shrink or grow depending on the size of the window rather than adjust the text location. 

### To Reproduce Bug

Go to any of the settings screens on mobile and it will let you scroll left and right. 

### Expected Behavior

You should not be able to scroll left and right. On desktop it should keep the font size the same and adjust the text location rather than shrinking it for smaller screens. 

### Install Method

Self-Compiled

### What version of WLED?

WLED_0.13.2-a0_ESP32

### Which microcontroller/board are you seeing the problem on?

ESP32

### Anything else?

Adding the following HTML fixes the issue. This line was on the settings_leds.htm so I copied and pasted it on the others and it fixed it. 

<meta content="width=device-width, initial-scale=1.0, maximum-scale=1.0, user-scalable=no" name="viewport">

This has been an issue for some time so it's not just the latest version.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2022-07-08:
Please checkout branch `blazoncek-merge-working` and test again.

FuriousGeorge91, 2022-07-08:
Yep it works as it should. Thank you!

closed 2022-11-01 by blazoncek