wled/WLED #2711: How do I cycle presets using a button

Bezur-Ro, 2022-07-10:
I have one button and I want that button to cycle through my presents that I've made so like you click it once and it goes to present 1 twice and preset 2 and so on is this possible?

amcewen, 2022-07-10:
I've got something similar part-written.  Rather than one click for preset 1; two clicks for preset 2; etc. it just cycles through each of the presets in turn when you click the button — then you can have lots of presets without getting too complicated.

There's a little video of it at the end of [this blog post](https://mcqn.com/posts/week-850-push-the-button/).

The code is [on this branch of my WLED fork](https://github.com/mcqn/wLED/tree/cycle-button-cycle).  It's a bit hacky at present, more of a proof-of-concept.  I need to...
* have it work out how many presets are available to cycle through (which shouldn't be too hard)
* work out how to turn it into a usermod or similar so it''s easier to include in the main code

I've not done much on it since the evenings got light and I've stopped needing my bike lights, but it'd be good to tidy up...

blazoncek, 2022-07-10:
Please use WLED [forum](https://wled.discourse.group) or Discord for help and support questions.
Please also check [KB](https://kno.wled.ge).
Closing.

closed 2022-07-10 by blazoncek