wled/WLED #2716: Inverted brightness values

DerSpatz, 2022-07-13:
I am currently designing small LED modules that can be used as light show for parties, raves etc. The modules are powered by 12V and use two WS2818 ICs. One WS2818 controls three RGB LEDs in series (because 12V), while the other WS2818 is used to control warm white, cold white and UV LEDs (3 each). As the WW, CW and UV LEDs have a much higher forward current (and I need the additional brightness), I am using INK1101 LED drivers (https://datasheet.lcsc.com/lcsc/2202132000_Cross-chip-INK1101ST_C2840169.pdf) for these LEDs. The DIM pin of the INK1101 works like an ENABLE pin and has an internal 20k pullup. Unfortunately, the output of the WS2818 is "active low", so I'd need an additional inverting circuit between WS2818 and the three INK1101s. Board space is limited, additional costs are bad (there's a lot of single modules, so a few cents matter), and I do not have a good power source for the inverting gate (the other ICs all have an input resistor on their positive supply), so I'd rather not try to add the inverters to the design.
The RGB data line and the WWCWUV data line are in parallel and independent from each other.

The easiest solution would be a software solution: Invert all the brightness values that are sent by WLED, so that internal brightness=0 is send out as brightness=255 and vice versa. 
This could be a simple checkbox in the settings, either for a certain output pin, a certain segment, or even for single LEDs in the LEDmap.json, maybe by putting a "!" in front of the inverted LEDs.

I assume this could also be useful for other hardware designs that use an external driver IC to run higher powered LEDs, and it could add interesting new twists to existing effects.

blazoncek, 2022-07-13:
Duplicate of #2574 
Closing.

closed 2022-07-13 by blazoncek

DerSpatz, 2022-07-13:
This is NOT a duplicate of #2574. #2574 is about inverting the output levels between MCU and LED.
In this issue the output levels between MCU and LEDs are not changed, instead, the values of the actual data is modified in a way that the output after the LED controller chip is inverted.

Please reopen!

blazoncek, 2022-07-13:
In this case it is not a feature of WLED but of NeoPixelBus which is the driver WLED uses for WS281x chipsets.
Please open an issue there.

DerSpatz, 2022-07-13:
Thank you. Will do!