wled/WLED #2717: Flashing segment

ShuffleBit, 2022-07-14:
### What happened?

Adding a black segment works, but that segment flashes white (LEDs and "Peek") about every 600ms or so.

I noticed this at 13.2-a0, so I reflashed with 13.1 to be on a release.

131 WS2812b LEDs, ESP8266 d1 mini, GPIO2, pushbutton on GPIO0, 15A brick, 470uF and 100nF caps across VIn and Gnd, 6cm data line length.

Whole string WLED effects are smooth and w/o blink/flashing. Whole string solid color works fine. The problem only occurs when a segment is defined via the web or via API calls.

### To Reproduce Bug

Set the whole string to white. Define a segment 1/3 of the string and set it black. Blink-blink-blink.

Click the "Peek" button and observe that the defined segment is sometimes white (and not the color set for the rest of the string!).

### Expected Behavior

The defined solid effect segment should not blink.

### Install Method

Binary from WLED.me

### What version of WLED?

0.13.1

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Anything else?

As always, this is a very cool project. You probably knew that.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2022-07-14:
You may have overlapping segments.

blazoncek, 2022-07-16:
You can close the issue if it is solved.

closed 2022-07-16 by ShuffleBit