wled/WLED #2718: Individual relay + analog/digital input for segments

drvkmr, 2022-07-17:
Hi all, big fan of WLED here and have been using it for projects for a while now. I am not a programmer but can still manage simple arduino code bits.

I live in a canal boat and have decided to connect all the lighting and controls to a single ESP32. Using WLED seemed like a good idea at the time I am struggling with making simple modifications to it now which are essential to my project.
Here's what the board looks like -
[image]
(There are 3 other boards which connect to this to enable long distance communication)

What I need here is the ability to control 8 individual LED segments via their dedicated analog/digital input + relay control for each of them. Mosfets are being used instead of relays but the code will be the same for both.

Now this is a big ask and I don't expect anyone to spend hours trying to make the perfect solution for me. What I will appreciate is some guidance around this so I can do this and potentially contribute to this amazing project in my own way.

I found a usermod called multi_relay but don't think it is designed for this case. I also understand the relay is normally driven by ON/OFF switch on maine wled page which makes this somewhat complex. But ideally this should be a part of webUI so it can be one of the settings within each segment.

Like I said before the way WLED works is too complex for me (thus the struggle) but will really appreciate if someone can guide me here!

blazoncek, 2022-07-17:
I wrote multi relay and it is well suited for your case.
When you power off WLED all relays are switched. If you need to switch individual relay when switching each segment you will need a special preset setup.

1st) You will need to mark every relay as EXTERNAL in usermod settings
2nd) create a preset for each segment with the content `{"seg":[{"id":X,"on":true}],"MultiRelay":[{"relay":X,"on":true}]}` for On state
3rd) create similar presets for Off state (replace `true` with `false`)
4th) use presets to toggle relays/segments on and off

For more, please use WLED [forum](https://wled.discourse.group) or Discord.
Closing.

closed 2022-07-17 by blazoncek

drvkmr, 2022-07-17:
Thanks, I will try that. And will post question about using analog input for each segment on discord !