wled/WLED #2720: Mapping multiple physical led to single logical led?

chrismelba, 2022-07-18:
For my led array I have a fan out pattern that would be great if I could have a single physical led at the start out to a dozen or more LEDs at the end. The examples only show one to one 

I could describe a map.json such as [0,1,1,2,2,2,3,3,4,4,4,4,5,6,6,6,]

This may already be possible? Or possible that segments are a better way to do this?

Love the project. Thank you so much for working so hard on it!

Thank you for your ideas for making WLED better!

blazoncek, 2022-07-18:
No, not possible with non-fixed-size setup.
Ledmap is used for mapping logical pixels to physical. What you want is map physical to logical. This is not and will not be implemented.
If you would have fixed-size set-up (i.e. 3 parallel strips) you could use ledmap and grouping in segment to overcome the problem but not when groups would change..

chrismelba, 2022-07-19:
Fixed size is fine. Can groups be different sizes to each other?

blazoncek, 2022-07-19:
only in different segments.

chrismelba, 2022-07-19:
I feel like I may not be explaining what I want quite right. I would like something similar to grouping, but instead of having every group be 2 LEDs I would like to have a variable number in each group?

chrismelba, 2022-07-19:
https://drive.google.com/file/d/1WOahsk68LSTV0kOtVMYsyOMiLuEmh9zc/view?usp=drivesdk

I've tried to explain better here