wled/WLED #2728: global brighness slider does not cause statechanged=true (autosave mod)

softhack007, 2022-07-26:
### What happened?

We received a bug report in SR WLED:
When using autosave usermod changing global brightness bar does not trigger autosave procedure. https://github.com/atuline/WLED/issues/218

We made a temporary fix in our brach, however the problem should also be  addresed in WLED. 

### To Reproduce Bug

Compile code with autosave enabled, change global brightness, wait 15 seconds, the mod does not save.

### Expected Behavior

We made a prliminary bugfix in our branch:

https://github.com/atuline/WLED/blob/6320ecc6c6d9d6f56773dd285b03c8a6be447726/wled00/led.cpp#L84-L91

[image]

Please check if the fix is doing the right thing, and consider to also fix the problem in WLED.

### Install Method

Self-Compiled

### What version of WLED?

WLED 0.13.2-a0 (main)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2022-07-27:
I do not think that would be a good idea.

softhack007, 2022-07-27:
What would be the good solution then?

All other changes (like effect sliders) lead to state changed, so the behaviour is at least inconsistent. 

See details in https://github.com/atuline/WLED/issues/218 and https://github.com/atuline/WLED/pull/219. You can ignore the part about "input level" which is SR specific.

softhack007, 2022-08-02:
Update: after some discussions with WLED people, we moved the preliminary fix out of 'applBri()' and into 'applyFinalBri()', which seems to be a little less dangerous.

````C++
void applyFinalBri() {
#ifdef USERMOD_AUTO_SAVE                  //WLEDSR - make autosave mod happy. A bit dangerous, as it may trigger unnecessary WS broadcasts (and other notificatios) 
  if (briOld != bri) stateChanged = true; //WLEDSR temporary fix, until solved in upstream
#endif
  briOld = bri;
  briT = bri;
  applyBri();
}
````