wled/WLED #2729: New Network- New issues- Wi-Fi dropping

balucanb, 2022-07-26:
Did not see my particular issue anywhere, Discord No help either. I do not think this is a bug. I have 7 WLED installations most running the latest build, I have had all of these pre-version 12 WLED. all running on a Dig-Uno or Dig-Quad (various integrations of those boards) I was having no issues until my ISP "upgraded" my network. All my lights are Wi-Fi, I have super connection strength and even when the problem happens if I look at what is /is not connected in the Eero app the lights show as being connected. What I have for for a router now is the Eero pro6 mesh system, 1 router and 4 beacons (their terminology) I was running a Google WiFi mesh before- no problems. The problem is the lights will not stay on network, they go from available to not all the time. All of them are running static IPs in the App and on my modem (ARRIS CM8200) Because of PPPP I needed to re flash most of the esp32's no issues there, I had copied all the code for my presets and just pasted those back in once I re flashed ( maybe that is not relevant, want to be thorough)  The obvious answer (to me) is the mesh system is somehow jacking this all up but I have no idea how to correct this and am looking for any help I can get.  The Eero is dual 2.4 5.0 not separate networks though. it does IPv6,UPnP, Thread and Client steering. You can temporally disable the 5.0  and disable Wi-Fi 6 but only on a temp basis in order to connect a older device, again maybe not of that is important but wanted to throw it out there. ANY CLUE how to remedy this? Thanks

blazoncek, 2022-07-27:
Get non-mesh access point and test. Mesh WiFi was reported to cause issues with WiFi stack on ESP.

balucanb, 2022-07-27:
No need to test I'm 99.9% sure that's it since before the new system it
working
I was hoping there might be a work around

balucanb, 2022-07-28:
Closing this. Eero does not seem to like WLED (haters) went back to my other Mesh system for now (Google) which works with zero issues for me.

closed 2022-07-28 by balucanb