wled/WLED #2730: wled not connecting to existing wifi network

rjworks, 2022-07-27:
### What happened?

using esp32 and used esptool to flash

ssid: telus
password: password
my wifi is 2.4G

static ip: 0.0.0.0 (default)
static gateway: 0.0.0.0 (default)
static subnet mask: 255.255.255.0 (default and same as my router)

disable wifi sleep: check

then when i click “Save & Connect”

it goes back to 4.3.2.1 homepage. i look at my the devices connected on my router and it’s not there. pls help

### To Reproduce Bug

using esp32 and used esptool to flash

ssid: telus
password: password
my wifi is 2.4G

static ip: 0.0.0.0 (default)
static gateway: 0.0.0.0 (default)
static subnet mask: 255.255.255.0 (default and same as my router)

disable wifi sleep: check

then when i click “Save & Connect”

it goes back to 4.3.2.1 homepage. i look at my the devices connected on my router and it’s not there. pls help

### Expected Behavior

connect to wifi

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.13.1

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2022-07-27:
Try "simple" password as a test. Some characters (in SSID and/or password) may cause issues.
If that doesn't help, get alternate access point and repeat.

rjworks, 2022-07-27:
> Try "simple" password as a test. Some characters (in SSID and/or password) may cause issues.
> If that doesn't help, get alternate access point and repeat.

thanks bro! what do you mean by get alternate access point?

balucanb, 2022-07-27:
Will try that

blazoncek, 2022-07-27:
> thanks bro! what do you mean by get alternate access point?

https://www.google.com/search?q=access+point

rjworks, 2022-07-27:
edit

i can connect to my phone hot spot but not router

rjworks, 2022-07-27:
what could be causing my router to act like this?

blazoncek, 2022-07-27:
Try disabling 5GHz, separating SSID for 2.4GHz and 5HGz, disable "advanced" features, etc.
Perhaps some newer features may not be properly implemented.

rjworks, 2022-07-27:
> Try disabling 5GHz, separating SSID for 2.4GHz and 5HGz, disable "advanced" features, etc.
> Perhaps some newer features may not be properly implemented.

i am already using 2.4g

softhack007, 2022-08-14:
Something that worked for me, when an ESP32 refuses to connect to wifi - especially for newly purchased devices: erase all flash, upload bootloader, upload fresh bin. Please remember that this procedure will erase all your settings, so better to make a backup first and restore settings later.

https://kno.wled.ge/basics/install-binary/

Flashing method 2: esptool

````cmd

esptool.py erase_flash

esptool.py write_flash 0x0 ./esp32_bootloader_v4.bin

esptool.py write_flash 0x10000 ./WLED_XXX.bin 

````

xumxum, 2022-08-22:
What board exactly it is? I just ordered two ESP32 Type-C devboards and after also struggling with the wifi I realized they are crap. The output power from them after flashing the wled is around -80, while my older devkit v1 with usb2 are around -25!! I also tried putting an esphome fw, which has some meaningful logs on the serial, you can see that it has good reception from the wifi, but always failing to connect with different errors..Even if it manages to connect, the ping can take from few seconds to miliseconds! So it seams this batch of devkits has very bad output wifi transmit power, maybe soldering or something on the TX..
I would advice to check the output power of the board when it's in access point mode with a tool like LinSSID or similar, if you don't get a good power output that would explain your issues..at least it does mine. I ordered 2 of this usb type C, thinking it's an upgrade... and both have the same issue..cheep is cheep

closed 2024-04-23 by github-actions[bot]