wled/WLED #2735: Support for the ESP32-S3

kylehovey, 2022-07-30:
**Is your feature request related to a problem? Please describe.**
No problem. :)

**Describe the solution you'd like**
I have multiple ESP32 S3 boards and would like to install WLED onto them. How hard would it be to configure the build for the S3?

**Describe alternatives you've considered**
I cloned the codebase and read through the existing configuration. I saw the config for the C3, but saw no config for the S3. I considered trying to add a config for the S3, but my knowledge in configuring the build parameters for this project is meager.

**Additional context**
I'm using the Adafruit QT Py ESP32 S3 board.

I'm happy to provide more context, just let me know. Thank you to all of the contributors for this amazing project!

blazoncek, 2022-07-30:
S2, S3 an C3 are not officially supported.
There were attempts by creating working environments but since not all libraries WLED uses are compatible with them we cannot guarantee it works.

softhack007, 2022-07-30:
Just for sake of completemess, the same applies for other  new ESP32 variants like ESP32-C2, ESP32-C6, ESP32-H2. Not officially  supported by WLED.

https://www.espressif.com/en/products/socs

kylehovey, 2022-07-30:
@blazoncek @softhack007 thank you, I'll go ahead and order some ESP32 boards in the mean time to stay in the supported board set. Out of curiosity, is there documentation on what libraries are not supported or a rough estimate of how much work it would take to support these new boards?

blazoncek, 2022-07-31:
No, but you can make one.
LittleFS was one of them but latest updates seemed to fix that.
Best is to to create an environment in PIO and start testing & debugging. Send your findings to library developers.

softhack007, 2022-08-02:
Update: Actually I just ordered an ESP32-S3, because it seems to be the only one of the new models with a realistic chance to run SoundReactive with many LEDs. 

However compiling a working binary for ESP32-S3 will probably become a "long march"... will post updates in case there are news.

Jason2866, 2022-08-06:
Imho the most important part is to do the changed Gpios first. After we had done this in Tasmota all other changes where not that complicated. RMT channels are less for the new MCUs! LittleFS is running stable in Tasmota on all variants. We use heavily.
Many libs are updated and do support the C3, S2 and S3 already.
Latest Tasmota core is now v2.4.0.1. Can be used with
```
platform = https://github.com/tasmota/platform-espressif32/releases/download/v2.0.4.1/platform-espressif32-2.0.4.1.zip
```

blazoncek, 2022-09-15:
Anyone interested may try https://github.com/blazoncek/WLED/tree/esp32-s2 
Support includes S2, S3 and C3

softhack007, 2022-09-17:
I use this in platformio_override.ini. Select environment `esp32s3dev_8MB`, compile & upload via USB port.
```ini
[esp32s3]
;; generic definitions for all ESP32-S3 boards
build_flags = -g
  -DESP32
  -DARDUINO_ARCH_ESP32
  -DARDUINO_ARCH_ESP32S3
  -DCONFIG_IDF_TARGET_ESP32S3
  -D CONFIG_ASYNC_TCP_USE_WDT=0
  -D CONFIG_LITTLEFS_FOR_IDF_3_2
  -D WLED_WATCHDOG_TIMEOUT=0
  -D ARDUINO_USB_MODE=1 -D ARDUINO_USB_CDC_ON_BOOT=0 -D ARDUINO_USB_MSC_ON_BOOT=0
  -DCO
lib_deps =
  ${env.lib_deps}
  ;; currently we need the latest NeoPixelBus dev version, because it contains important bugfixes for -S3
  https://github.com/Makuna/NeoPixelBus.git#master @ 2.7.0
  https://github.com/pbolduc/AsyncTCP.git @ 1.2.0

[env:esp32s3dev_8MB]
;; "standard" ESP32-S3 development board, with 8MB FLASH, no PSRAM
board = esp32-c3-devkitm-1
framework = arduino

platform = espressif32@^5.1.1
platform_packages = platformio/framework-arduinoespressif32@^3.20004.220825

upload_speed = 921600
build_unflags = ${common.build_unflags}
build_flags =  ${common.build_flags} ${esp32s3.build_flags} -D WLED_RELEASE_NAME=ESP32S3 #-D WLED_DISABLE_BLYNK #-D WLED_DISABLE_BROWNOUT_DET
lib_deps = ${esp32s3.lib_deps}

board_build.partitions = tools/WLED_ESP32_8MB.csv
; board_build.flash_mode = dio
board_build.f_flash = 80000000L
board_build.flash_mode = qio
; board_build.f_flash = 120000000L
monitor_filters = esp32_exception_decoder

```

softhack007, 2022-09-17:
If you want something to play with, this is an **experimental build** of https://github.com/blazoncek/WLED/tree/esp32-s2 for ESP32-S3 with 8MB FLASH:

1. `esptool erase_flash`
2. `esptool write_flash 0x0 esp32S3_bootloader_qio_80m_arduino204.bin`
3. `esptool write_flash 0x00010000 WLED_esp32s3dev_8MB_experimental_debug_firmware.bin`

Files: 
* [esp32S3_bootloader_qio_80m_arduino204.zip](https://github.com/Aircoookie/WLED/files/9591758/esp32S3_bootloader_qio_80m_arduino204.zip)
* [WLED_esp32s3dev_8MB_experimental_debug_firmware.zip](https://github.com/Aircoookie/WLED/files/9591779/WLED_esp32s3dev_8MB_experimental_debug_firmware.zip)

blazoncek, 2022-09-18:
Sometimes it may be necessary to update boot partition table as well. (At offset 0x8000)

softhack007, 2022-09-18:
@blazoncek good point, did not think of that 👍 Is it the "partitions.bin" that I find in the .pio build folder?

btw, my boot log starts like this.  Not sure what these extra "load" messages and "SHA-256 comparison failed" means ...
```ShellSession
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x9 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x43c
load:0x403c9700,len:0xbec
load:0x403cc700,len:0x2a3c
SHA-256 comparison failed:
Calculated: dcde8d8a4817d9bf5d5d69a7247667264e4e10ac7493514868b61f5aa6146539
Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
Attempting to boot anyway...
entry 0x403c98d8

---WLED 0.14.0-bl0 2209172 INIT---
esp32 v4.4.1-472-gc9140caf8c
arduino-esp32 v2.0.4
CPU:   ESP32-S3 rev.0, 2 core(s), 240MHz.
FLASH: 8MB, Mode 2 (DIO), speed 80MHz.
```

softhack007, 2022-09-18:
PS: the "file editing bug" is also described here: https://github.com/me-no-dev/ESPAsyncWebServer/issues/1210

blazoncek, 2022-09-18:
> @blazoncek good point, did not think of that 👍 Is it the "partitions.bin" that I find in the .pio build folder?
Yes.

closed 2022-10-19 by Aircoookie

softhack007, 2023-01-25:
As of today, the best approach for -S3 is still to build & upload your own firmware yourself in platformio. Pre-build .bin files are not working everywhere.

Make sure you __don't__ flash the standard "ESP32 V4 bootloader" from the WLED release page. It will not run, as the -S3 has a different instruction set. You need a special bootloader for -S3. Build & Upload in platformio will take care of that.

softhack007, 2023-03-30:
> To all users of the S3: Are there any major issues with the S3?

Well, it depends. All new MCUs (-S3/-S2/-C3) are still **bleeding edge experimental**. From technical perspective, the -S3 might be a good choice (compared to -S2 or -C3) as its performance (speed) is similar to "classic ESP32". Also it does support PDM and I2S microphones, while -C3 and -S2 lack PDM support.

Personally I have one -S3 board without PSRAM, and it works ok for me. But I know others have -S3 boards where WLED would not even boot up.

I can recall a few "limitations" with the new MCUs (not only for -S3), maybe there are more:
* audioreactive: no support for analog microphones
* I2C does not work well (may hang your device)
* Serial sometimes hangs, then the chip heats up, and you need to erase flash to make it work again
* bootloader: it can be a nightmare to find the "right" bootloader for each board. "classic esp32" bootloaders will no work.
* ESP32 supports up to 10 LED outputs, but the max on -S3 is 4 outputs (3 on -S2, 2 on -C3).

> I would like to use it for a sound reactive project and create a custom board for it.

As always, it's your call to see if that can be done. Maybe build a prototype, then test it to death. Then decide if what you have a good enough for a project. Start small, and be prepared that you might come to a point where you find that -S3 is not good enough for what you want to achieve. So your backup plan should be to use "classic ESP32" like the pico-D4 (dual core).

blazoncek, 2023-03-30:
> To all users of the S3: Are there any major issues with the S3?

Next time, please, use WLED [forum](https://wled.discourse.group) or Discord for help and support questions.

softhack007, 2023-04-27:
Talking about this one: https://www.wemos.cc/en/latest/s3/s3_pro.html ?

> WEMOS - LOLIN S3 PRO: >>>>> NO I2S <<<<<<

Strange ... the -S3 has two I2S units, the only thing that's not possible is I2S-ADC for analog mics or analog line-in.
Could be that you'd need to make your own custom build for this unit, to get a firmware with the right memory modes. 

Memory modes are a nightmare on -S3, and firmware build for one specific mode will not run on other boards that have a different mode.

githubmoose, 2023-04-27:
@Softhack007: that is the most frustrating part, I know they áre in but not routed to the outside. One can indeed create some ‘pseudo-I2S out’ in fw but I have enough other challenges to spend my time on. 
Having said that, there are currently things going on to the east of where I live that make things like this again quite insignificant…..

Fuxy22, 2024-06-28:
Dammit! Just got a "Arduino Nano ESP32" with a ESP32-S3 thinking any ESP32 chip would work... any plans to support this in the future? I foresee this becoming a really popular chip/pitfall.

Well such is life... i'l go bash my head against a wall then...

blazoncek, 2024-06-29:
The S3 is supported as far as WLED code goes. You may need to fiddle with flashing process, but that's all. @Fuxy22 

As for I2S, it has 2, but Espressif decided to change API and it is incompatible with NeoPixelBus (as of this writing), so WLED can only use 4 RMT channels for LED output. Audio works though, kind of.

CYber-Dinium, 2024-12-07:
Those ovoeride ini.
I'm new songoneasy.
How do I use them?

And how to flash for esp32 s3 16 8mb
Esl32 s3 8mb 4mb vroom variations.

DedeHai, 2024-12-08:
@CYber-Dinium github is for tracking issues, please use the forum or discord for support questions.