wled/WLED #2743: Toggle and Brightness in same command won't toggle

matthias-steinkamp, 2022-08-08:
### What happened?

If you use brigthness and on=t, the LEDs will be off (turned on by brightness and turned of by toggle)

### To Reproduce Bug

use this json command {"on":"t", "bri":128}

### Expected Behavior

brightness is set to 128
LEDs on state is toggled, depending on their former power state

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.13.0-b6 (2112080)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

dosipod, 2022-08-08:
Could you double check the version you are using .  WLED 0.13.0-b6   is the sound reactive fork  ?

blazoncek, 2022-08-08:
> Could you double check the version you are using . WLED 0.13.0-b6 is the sound reactive fork ?

The error is confirmed in source. The solution is something like this:
```c++
  if (!root["on"].isNull()) {
    bool on = root["on"] | onBefore;
    if (on != onBefore || (root["on"].is<const char*>() && root["on"].as<const char*>()[0] == 't')) toggleOnOff();
  } else {
    if ((onBefore && bri==0) || (!onBefore && bri>0)) toggleOnOff();
  }
```

dosipod, 2022-08-15:
We are trying to use an  MCU for control as a simple smart switch without HA .But  turning off received  brightness will  not work ( unit  will not receive on /off ) . I am assuming that is the same case above  as ON/OFF  and brightness  is joined

blazoncek, 2022-08-18:
Fixed in #2737

closed 2022-08-18 by blazoncek