wled/WLED #2750: Random flashing after upgrading to 0.13.2

Plawasan, 2022-08-15:
### What happened?

Upgraded to 0.13.2 (official ESP32 binary), now randomly every ~20-60s the entire strip flashes briefly regardless of preset used. Downgrading to 0.13.1 fixes the issue. I've been using this HW setup (ESP32/60xWS2812) for several months now without any issues whatsoever.

Video of the issue (should be solid blue) - filmed in slow-mo, in reality the flash lasts ~300ms by my estimate.

https://www.youtube.com/watch?v=eUHhXFYiDpY

### To Reproduce Bug

Upgrade to 0.13.2

### Expected Behavior

Do not flash...

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.13.2 (build 2208140)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

berendhaan, 2022-08-15:
I have the same issue.

Environment: esp32 v3.3.6-16-gcc5440f6a2 (0)

Bloodpack, 2022-08-15:
I have the same issue

swampylee, 2022-08-15:
Same issue here on all my esp32’s.

madmic1314, 2022-08-15:
Same issue here as well - also when using clock mode on a 60 led ring.

Home Assistant seems to show the device going offline, though not dropping ping or off wireless.

knd775, 2022-08-15:
Also happening here.

Some info in case it helps:
Strong wifi signal (~54dBm)
214 x `WS2812B` LEDs
Not a power supply issue

`0.13.2 (2208140)`
`esp32 v3.3.6-16-gcc5440f6a2`

blazoncek, 2022-08-15:
Are you using analog "buttons"?

If not the only real change was the addition of WDT.
@Aircoookie perhaps a build with -D WLED_WATCHDOG_TIMEOUT=0?

Pomezny, 2022-08-15:
I am not sure yet how the watchdog works, but after the "glitch" uptime on my ESP32 does not reset. Right now I have around 30 minutes uptime according to the "info tab" and noticed flash multiple times.

meierthered, 2022-08-15:
I am also having this issue

LarsBorris, 2022-08-15:
I am also having this issue

Pomezny, 2022-08-15:
Well, it looks I was wrong and @blazoncek was correct. Tried building with WLED_WATCHDOG_TIMEOUT set to 0 and it looks like flickering disappeared. Even when changing brightness (as mentioned in my second EDIT of previous comment).

Aircoookie, 2022-08-15:
Could confirm your findings @Pomezny with the "brightness slider test" (we'll definitely want to add rate limiting there, I wasn't aware arrow keys trigger `onchange`).
I fail to see any flickering in regular use however even with the watchdog enabled. Still, given the feedback, it might be better to disable it again by default for the time being.

Here is the ESP32 0.13.2 binary without WDT enabled for testing:
https://drive.google.com/file/d/1ebxrBeNOs8VfspFOIBmA3v-uE--e-lev/view?usp=sharing

Bloodpack, 2022-08-16:
@Aircoookie 

Flashed your modified firmware and flickering is gone.

ESP32 Nodemcu

Thank you

blazoncek, 2022-08-16:
There is another option.
Add this to the build environment for ESP32:
```
board_build.f_flash = 80000000L
board_build.flash_mode = qio
```

swampylee, 2022-08-16:
> Could confirm your findings @Pomezny with the "brightness slider test" (we'll definitely want to add rate limiting there, I wasn't aware arrow keys trigger `onchange`). I fail to see any flickering [...]

Thank you, fixed for me too.

Plawasan, 2022-08-16:
> Thank you, fixed for me too.

Fixed for me as well, feel free to close the issue.

LarsBorris, 2022-08-16:
> Could confirm your findings @Pomezny with the "brightness slider test" (we'll definitely want to add rate limiting there, I wasn't aware arrow keys trigger `onchange`). I fail to see any flickering [...]

I just flashed the new binary - testing but so far looks good!

Phara0h, 2022-08-17:
Same issue I was chasing hardware issues and then I noticed it's happening on all my strips xD. Rip 5 hours

LarsBorris, 2022-08-17:
> I just flashed the new binary - testing but so far looks good!

I can confirm that the problem is gone.

thewolf-oz, 2022-08-17:
I also flashed the new binary and flickering is gone

hell1337hound, 2022-08-17:
Same issue here with 3 installs at home.

poelzi, 2022-08-18:
This platform just confuses me. I tested the watchdog on multiple controllers here and did not notice such behaviour. I often have unreachable ESPs here which added the watchdog fixed. I start to suspect that certain wlan configurations might cause the deadlock.
The watchdog is used in esphome, so I don't think it is broken in general. Which GPIO ports do you use ? How many ?

blazoncek, 2022-08-18:
@poelzi I tested WDT for over two weeks before deciding to merge it into master. While I did notice occasional random LED blink (more so during UI access) I attributed it to my test set-up which contains a lot of dupont wires all over the place. I never thought *feeding watch dog* would be the source of it. 😄 

I use GPIOs: 16, 3, 26 & 5, the latter two for CCT analog strip. The flicker was also present on a single GPIO 16 (WROOM) or GPIO 27 (WROVER model). That being said it was just an occasional (about 10s or more between) flash of LED or a few LEDs at a time.

When I added a proposed solution above to increase flash speed and double the channels for it, flickering disappeared completely.

Pomezny, 2022-08-18:
@poelzi I use only a single strip of 94 LEDs and I have it added to Home assistant. As I was curious, I flashed back the version with WTD and from my observations its kinda sporadic, but when flashing starts it never ends and flashes like every minute. I am not familiar with ESP32 and I suspected CPU spending too much time in interrupts handling (if this even is interrupt driven). That may explain why it is sporadic - different network traffic for instance.  BUT when I try moving the brightness slider as described above, the flash is sometimes very fast. I would expect reboot loop at period <3s, even in case of total resource starvation reboot  should not happen sooner than 3s timeout. If I understand it correctly. I even added esp_task_wdt_reset right after esp_task_wdt_add to ensure that WDT is fed at least once after reboot and still, I get flashes faster than 3s. Have a look at the attached video. That boggles my mind. Flicker is not present with WDT off, but if the flash is reboot due to WDT how it can reboot faster then 3s?

https://user-images.githubusercontent.com/4263174/185435077-f8681ffe-69c2-4022-95fb-53c9b807064e.mp4

Pomezny, 2022-08-18:
A little update. I just did a quick and dirty test. Added variable and condition to jump into endless loop with yield()(yield should prevent WDT on idle task firing, results are identical with or without it) after 10s of uptime. (yield should prevent WDT on idle task firing, results are identical with or without it)
When WDT fires and reboots device it will:
1) power off the LEDs. Basically my LED strip turns off afters 10s.
2) uptime in info tab resets.
According to this, it looks to me, that flashing is actually not ESP rebooting itself because WDT is not fed in time. Obviously it has something to do with the #2657, since turning off the WDT fixes the issue, but I have no idea what is the side effect causing the flashing. At the time I don't have enough free time to dive into project this big (and written in C) and resolve root of the cause. But if there is something I can check or try just tag me and I will try my best to help this awesome project.

screech24, 2022-08-19:
Just wanted to comment and say for the record I am also having this issue after updating to 0.13.2

fenty17, 2022-08-20:
Also been having this issue. Is a fix going to be released soon? If not I'll also try the file that's worked for others.

blazoncek, 2022-08-20:
> Also been having this issue. Is a fix going to be released soon? If not I'll also try the file that's worked for others.

That is the official fix.

jerobins, 2022-08-20:
But it is not released; hoping for a 0.13.3 release to trigger the update
mechanisms in place...

PDekker, 2022-08-20:
> 

Will I receive a notification for a OTA update for this?

Aircoookie, 2022-08-23:
0.13.3 is now released, fixing this issue. Please reopen in case the issue persists after updating :)

closed 2022-08-23 by Aircoookie