wled/WLED #2751: Restart mit der Fritzbox beta

PatricckW, 2022-08-15:
### What happened?

Guten Tag,
Ich habe bei Codm.de den Pixel

controller mit dem ESP82 gekauft.

Nun habe ich das Problem das der mit der Aktuellen Beta 7,39-98869 auf der Fritzbox 7590 das der Controller sobald er den ersten Reset (Hängt an eine Repeater) bekommt genau nach 60 Sec neu startet. 
Mit der aktuellen Release Version der Box macht er das nicht.

Grüße Patrick Wörmbke
[image]
[image]

### To Reproduce Bug

Mit der aktuellen Beta Version 7.39-98869 nach dem Update auf diese Version und einmal neustart des Controllers macht er alle 60 Sec einen Restart 

### Expected Behavior

Install the beta on a 7590

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.13.1

### Which microcontroller/board are you seeing the problem on?

ESP8266, ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

softhack007, 2022-08-15:
Hi Patrick,

Unfortunately most of the people here (developers, testers, etc) are not speaking German. Also google translate is not help with technical texts. Could you provide a summary of your problem in englisch, please?

Also I am not sure if you describe a problem with your FritzBox (DSL WiFi router), or is the problem related to WLED software running or your ESP8266?  Please try if the problem still exists when you "downgrade" the router to the latest official FritzOS release ( = not a BETA version).

PatricckW, 2022-08-15:
I downgrade it, and all fine, but i think what happened if AVM released a new version and many users have this problem. I wondering why it reboot after 60 seconds, and I figured out it was with the beta version of the Internet router. Von meinem Telefon gesendet-------- Ursprüngliche Nachricht --------Von: Frank ***@***.***>Datum: Mo., 15. Aug. 2022, 16:39An: Aircoookie/WLED ***@***.***>Cc: PatricckW ***@***.***>, Author ***@***.***>Betreff: Re: [Aircoookie/WLED] Restart mit der Fritzbox beta (Issue #2751)
Hi Patrick,
Unfortunately most of the people here (developers, testers, etc) are not speaking German. Also google translate is not help with technical texts. Could you provide a summary of your problem in englisch, please?
Also I am not sure if you describe a problem with your FritzBox (DSL WiFi router), or is the problem related to WLED software running or your ESP8266?  Please try if the problem still exists when you "downgrade" the router to the latest official FritzOS release ( = not a BETA version).

closed 2022-11-01 by blazoncek