wled/WLED #2753: Remote Buttons utilising ESPNow

jonwilliams84, 2022-08-18:
Would it be possible to add remote button(s) using ESPNow?

The main wled ESP (attached to LEDs) would be the receiver; you would configure a sender or multiple sender ESP boards, along with button mappings. E.g. When Sender1 Button1 is pressed toggle on/off. When Sender1 Button2 is pressed go to next preset/effect, and so on and so forth.

The Sender(s) could allow you to hook up 1 or more PULL_UP/PULL_DOWN buttons that map to the buttons defined in the main wled config. The unit could then enter DeepSleep after sending the ESPNow message, allowing the sender to run on batteries for a long time.

**Describe alternatives you've considered**
I have explored having a separate receiver with ESPNow to MQTT, to listen for message sent from senders and act as an MQTT bridge to trigger an automation in HomeAssistant that turns on WLED/Toggles Effect. However this is probably a bit overkill and adds requirement for MQTT and an additional ESP unit to be on and listening for messages to relay.

blazoncek, 2022-11-18:
Not in any immediate plans.

blazoncek, 2023-09-19:
Implemented via #3237

closed 2023-09-19 by blazoncek

insane2subro, 2024-07-10:
I made a small ESPNOW to MQTT gateway with support for WLED. You can send a MQTT payload to the ESP with WizButton number and it will publish ESPNOW messages.
https://github.com/insane2subro/ESPNOW2MQTT