wled/WLED #2755: Unknown character in presets.json  - homeassistant integration

adplante, 2022-08-18:
### What happened?

"Unknown Error' when setting up WLED in Homeassistant.  

References character 1270 in presets.json which is an "&", the very last character in the file.  

Presets.json attached in logs. 

See Frenck's response:
https://github.com/home-assistant/core/issues/76772

WLED 0.13.1
Dig Uno ESP32
Flashed via install.wled.me
Homeassistant v 2022.8.6

### To Reproduce Bug

Unknown Error in Homeassistant when setting up a new integration. 

### Expected Behavior

Integration into Homeassistant

### Install Method

Binary from WLED.me

### What version of WLED?

0.13.1

### Which microcontroller/board are you seeing the problem on?

ESP32

### Relevant log/trace output

```shell
{"0":{}                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                 ,"1":{"n":"1","ql":"1","on":true,"bri":155,"transition":7,"mainseg":0,"seg":[{"id":0,"start":0,"stop":210,"grp":1,"spc":0,"of":0,"on":true,"frz":false,"bri":255,"cct":127,"col":[[255,168,84,214],[255,0,0,255],[247,13,255,255]],"fx":0,"sx":128,"ix":128,"pal":2,"sel":true,"rev":false,"mi":false},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0}]}&
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

dosipod, 2022-08-18:
Any body having the issue to test deleting the preset.json and then create a new preset to see if that fix the issue .  
[image]
we could not replicate as we do not have the same error but that is a basic test

blazoncek, 2022-08-18:
As a workaround remove the extraneous `&` at the end of file using the /edit page and save the change.

The error may be caused by unexpected interrupt occurring during file write operations.
This is being investigated but so far has not produced a reliable reproducible method.

craigthompson, 2022-09-29:
I have something similar in that using Home Assistant sometimes the presets file gets corrupted.

blazoncek, 2022-10-10:
Please follow up on #2793 
Closing.

closed 2022-10-10 by blazoncek