wled/WLED #2756: No AP on ESP32 with current (separate) bootloader

alxndr42, 2022-08-18:
### What happened?

I tried flashing WLED 0.13.2 and the separate V4 bootloader onto [this ESP32 board](https://www.amazon.de/gp/product/B08TBPBJGV), using esptool, according to the [website instructions](https://kno.wled.ge/basics/install-binary/#esp32), which seemed to work, but the AP did not appear.

What worked instead was `esptool.py write_flash 0x0 WLED_0.12.0_ESP32_with_bootloader.bin` and updating to 0.13.2 via the Web UI.

### To Reproduce Bug

See above

### Expected Behavior

Flashing the firmware and bootloader should result in a functional controller

### Install Method

Binary from WLED.me

### What version of WLED?

0.13.2

### Which microcontroller/board are you seeing the problem on?

ESP32

### Relevant log/trace output

```shell
Output of "esptool.py chip_id":

esptool.py v4.2.1
Found 3 serial ports
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting....
Detecting chip type... ESP32
Chip is ESP32-D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 78:e3:6d:XX:XX:XX
Uploading stub...
Running stub...
Stub running...
Warning: ESP32 has no Chip ID. Reading MAC instead.
MAC: 78:e3:6d:XX:XX:XX
Hard resetting via RTS pin...
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

dosipod, 2022-08-20:
Below commands worked on multiple esp32 with the latest bin  
https://github.com/Aircoookie/WLED/releases/download/v0.13.2/WLED_0.13.2_ESP32.bin 
We can see the units AP and connect to it and configure wled , save presets ..etc 

As the question was limited to the AP appearing ,we did not do any extensive functional tests but units seems stable 

esptool.exe -p COM9 erase_flash
esptool.exe -p COM9 write_flash 0x0 esp32_bootloader_v4.bin
esptool.exe -p COM9 write_flash 0x010000 WLED_0.13.2_ESP32.bin

alxndr42, 2022-08-20:
@dosipod Like I wrote in the issue description, I also followed the official flashing instructions. But they didn't work on my ESP32, hence this issue.

alxndr42, 2022-08-20:
I did some systematic testing to see if this was related to the bootloader version or something else, the results:

No AP

```
esptool.py erase_flash
esptool.py write_flash 0x0 ./esp32_bootloader_v4.bin
esptool.py write_flash 0x10000 ./WLED_0.13.2_ESP32.bin
```

No AP

```
esptool.py erase_flash
esptool.py write_flash 0x0 ./esp32_bootloader_v2.bin
esptool.py write_flash 0x10000 ./WLED_0.13.2_ESP32.bin
```

AP works

```
esptool.py erase_flash
esptool.py write_flash 0x0 ./esp32_bootloader_v2.bin
esptool.py write_flash 0x10000 ./WLED_0.12.0_ESP32.bin
```

AP works

```
esptool.py erase_flash
esptool.py write_flash 0x0 ./esp32_bootloader_v4.bin
esptool.py write_flash 0x10000 ./WLED_0.12.0_ESP32.bin
```

AP works

```
esptool.py erase_flash
esptool.py write_flash 0x0 ./WLED_0.12.0_ESP32_with_bootloader.bin
```

AP works

```
esptool.py erase_flash
esptool.py write_flash 0x0 ./WLED_0.12.0_ESP32_with_bootloader.bin
esptool.py write_flash 0x0 ./esp32_bootloader_v4.bin
```

AP works

```
esptool.py erase_flash
esptool.py write_flash 0x0 ./WLED_0.12.0_ESP32_with_bootloader.bin
esptool.py write_flash 0x10000 ./WLED_0.13.2_ESP32.bin
```

AP works

```
esptool.py erase_flash
esptool.py write_flash 0x0 ./esp32_bootloader_v4.bin
esptool.py write_flash 0x10000 ./WLED_0.12.0_ESP32.bin
esptool.py write_flash 0x10000 ./WLED_0.13.2_ESP32.bin
```

So it looks like v0.13 will only work on this particular ESP32 if v0.12 was running previously. Very curious.

blazoncek, 2022-08-20:
Why not just use the variant that works for you and then update OTA?

alxndr42, 2022-08-20:
@blazoncek I don't understand the question, TBH. There's some kind of bug when installing the latest release (instead of upgrading). But instead of filing an issue, I am supposed to find an undocumented workaround, then keep quiet about it?

blazoncek, 2022-08-20:
I never said anything of such sort. What are you implying?
I was merely curious why not use something that works.

alxndr42, 2022-08-20:
@blazoncek Well, where did you get the impression that I'm not using "something that  works"? Especially after I describe, in detail, how to arrive at "something that works" in this issue.

In the spirit of improving this open source project, I thought it might be useful to share this knowledge. So, you know, the developers can benefit from it, or any other users running into the same problem.

blazoncek, 2022-08-20:
Did you monitor serial output? What did it say *after* flashing when ESP was booting? **That** would help, more than showing which combination worked for you.
You can also request debug build with extensive logging to serial to help pinpoint the cause of error.
You can also build from source and try to upload *your* binary to verify if the binary *may be the culprit* (I have seen corrupt binaries in the past).
Then you can also try to use binaries from @srg74 as he provides them bundled with various usermods included and see if those also exhibit similar behaviour.

alxndr42, 2022-08-20:
@blazoncek You're actually beginning to sound helpful now. But I disagree, the results I posted indicate there is clearly something wrong with a certain type of installation. A developer might be able to suggest a better starting point for further investigation based on this, rather than me randomly trying to install other builds. I've been using the binaries hosted on GitHub BTW, I really hope those aren't corrupt.

And no, I didn't monitor serial output (don't know how) or build the software myself, I'm just a WLED user trying to help.

blazoncek, 2022-08-20:
Try a binary from here: https://github.com/Aircoookie/WLED/issues/2750#issuecomment-1215970151

blazoncek, 2022-08-20:
That one should work.
If you would have monitored serial output and posted it I could have answered immediately.

To monitor serial output you do not need to be programmer or anything (though if you are using esptool you are no beginner IMO). All you need to do is attach ESP using USB cable to your computer (mot likely already done if using esptool) and a terminal application. Any one that can display COM port data will do but some prefer Arduino IDE and some MS VSC+PIO.

The problem you are seeing is due to LittleFS not being initialised and WDT kicking in during formatting of FS.
I think you will be able to close this issue since the culprit is known. @Aircoookie may release updated 0.13.2 (or perhaps renumbered to 0.13.3 to reduce confusion) soon.

alxndr42, 2022-08-21:
@blazoncek And how am I supposed to know all this after following the installation instructions?

I didn't come here for this type of behaviour, so I am blocking you now (please stop interacting with me in any way), and will let the WLED developers use the information I provided as they see fit.

closed 2022-08-21 by alxndr42

blazoncek, 2022-08-21:
Dear @alxndr42 but I am a WLED developer, the one responsible for all the bells and whistles of the upcoming 0.14. And the one responsible for the blunder with WDT in 0.13.2.
You should've checked my status if you want to complain or try to insult. Boy, you act alike a teenager/child.
🤣

Aircoookie, 2022-08-23:
0.13.3 is now released, fixing this issue :)

alxndr42, 2022-08-24:
@Aircoookie Awesome, thank you!

I can confirm the AP works on my particular ESP32 using these commands:

```
esptool.py erase_flash
esptool.py write_flash 0x0 ./esp32_bootloader_v4.bin
esptool.py write_flash 0x10000 ./WLED_0.13.3_ESP32.bin
```