wled/WLED #2761: 13.3 OTA on ESP3266 - Unresponsive

flyfishUtah, 2022-08-23:
### What happened?

I did an Manual OTA update via the IP address UI of the ESP3266.  Downlaoded the ESP3266 and uploaded, and it said it was successful and rebooting.   Never did come onlline and connect to the network.  The blue light is not on with the device.  A dozen or so of the lights are on (white) with the final light being red.   Has worked flawlesssly for a few years.   Was on 13.1

### To Reproduce Bug

Manual OTA 3266.bin file uploaded via the UI, said it was successful, and was restarting.  IP address is not reachable (reserved DHCP), no blue light, with some lights on.

### Expected Behavior

UI render and device reachable

### Install Method

Binary from WLED.me

### What version of WLED?

0.13.1 upgrade to 0.13.3

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

AlexKusnezov, 2022-08-23:
same here, two of my eleven wled devices misbehaved after I updated them via home assistant. The esp32 froze every 5-10 minutes and the esp8266 one every half an hour when it was turned on. I solved the issue on esp32 by reberting to 0.13.3.2. Unfortunately revelting didn't solve the issue for the esp8266 one

mattbruman, 2022-08-23:
I had 3 devices that became unresponsive as well. I flashed from 13.2 to 13.3 through Home assistant and they never woke up. 2 were d1 mini pro and 1 was d1 mini classic. I manually flashed them with 13.3 and things have been running well after.

flyfishUtah, 2022-08-24:
I ended up having to manually flash via Esphome flasher to the ESP8266, and finally got 0.13.3 to work; HOWEVER, when I tried to install my back up configs, it said it was successful, but nothing applied.  Presets backup worked perfectly.

Did the same thing on my ESP32 and flashed via (via web) and both backup files applied succesfully.

Something is not working well with the ESP8266 image

blazoncek, 2022-08-24:
After uploading cfg.json immediately reboot ESP.
This will be fixed in 0.14

AlexKusnezov, 2022-08-24:
I've ended up reverting the two affected nodemcu esp9266 and nodemcu esp32 tp 13.2 and the restarting home assistant. They work now. The only thing they have both in common are playlists. The other nine deives didn't have any and the update on those went well. Maybe there's something with the playlists.

blazoncek, 2022-08-24:
The only difference between 0.13.2 and 0.13.3 is added support for LPD6803 LEDs and disabled WDT. WDT was causing boot and flickering issues on ESP32.

thommy86, 2022-09-22:
How to solve this issue? I have an Athom bulb flashed ota with the same issue. Is there a way to connect ota? Do I need to open up the bulb?

blazoncek, 2022-11-18:
Is this still an issue?

flyfishUtah, 2022-11-18:
> Is this still an issue?

I reflashed to resolve it.  So not an issue for me know.

PiotrMachowski, 2022-11-18:
I also had this problem and needed to reflash the esp

blazoncek, 2022-11-19:
I am going to close as it was not possible to reproduce and re-flashing solved the issue.
It may have been random configuration issue.

closed 2022-11-19 by blazoncek