wled/WLED #2764: How to prevent Relay triggering on boot or reset

PixelLights, 2022-08-23:
### What happened?

I have a relay module board which uses optocouplers, and has an active low.  On boot, the relay triggers momentarily which causes the connected device to come on which is unwanted behaviour. I'm using standard relay mode, but I'll try with USERMOD_MULTI_RELAY too to se if the problem occurs. How to solve this problem ?

I am using the WT32 - ETH01 board with the ethernet active. Relay tested on pin 2 and 4, with the same issue. 

### To Reproduce Bug

Just connect the relay board with an active low on the relay pin.

### Expected Behavior

The relay should stay off when starting WLED.

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.13.2

### Which microcontroller/board are you seeing the problem on?

Other

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PixelLights, 2022-08-24:
If I use USERMOD_MULTI_RELAY and configure pin 4 for the relay 0 the same behavior occurs. But if I check "relay-0 external" the problem stops, but it will only switch through the API

blazoncek, 2022-08-24:
Have you tried another GPIO? Some GPIO pins output something during boot.
You can also uncheck "active-high" and reverse contacts on relay.

majdzik84, 2022-08-24:
In version 0.11 everything worked fine for me. After updating to 0.13 I have the same problem. When voltage is applied, the LEDs (relay) turn on. I found that there is no such concern when the boot preset is set to 0. .[image]
Unfortunately I would like to have a boot preset defined and hope this gets fixed

blazoncek, 2022-08-25:
I have about 20 devices set up with relay some with boot preset some without and I am not seeing this behaviour.
Please post your cfg.json and wiring schematics and a picture of your relay module.

closed 2022-11-01 by blazoncek