wled/WLED #2765: Feature request: cycle brightness with API command

mistic100, 2022-08-24:
**Is your feature request related to a problem? Please describe.**
When using a preset with the API command `&A=~+` one can have a button that, while pressed, progressively increase the brightness then go back to 0 when reaching 255.

Problem is increments of 1 are painfully slow (it looks like the preset is executed every 1s while keeping the button pressed, so more than 4 minutes to cycle through the whole brightness range !)

And another problem is that when using the command `&A=~10` the brightness will not cycle but clip to 255.

**Describe the solution you'd like**
I know that this behaviour is clearly documented, but I think it is rather limiting, not only for brightness but for other parameters that support `~`.

So my request is to not clip anymore but always wrap, whereas `~+` or `~x` is used (it would also be more consistent).

**Describe alternatives you've considered**
Having another preset `&A=10` executed on double-press. But I am designing a one-button system and was planing to configure

- single press : `&T=2`
- long press : `&A=~10`
- double press : `&PL=~`

blazoncek, 2022-08-25:
You can use button 1 for this behaviour without assigning presets.
Use default behaviour.

mistic100, 2022-08-25:
Okay that's fine for my case. For anyone finding this, excerpt from https://kno.wled.ge/features/macros/

> The default (built-in) actions for button 0 are short-press: toggle on/off and long-press: select random color. Long-pressing for more than 6 seconds will open the WLED-AP with the default password (wled1234).
For further buttons, the default action for short press is cycling effects, long press ramp brightness, and double press cycle palettes.

@blazoncek I still think the API would benefits such changes for advanced cases (not only A but other parameters). If it's okay I can propose a PR, otherwise feel free to close the issue.

Thank you

blazoncek, 2022-08-25:
There are two opposing wishes for this kind of behaviour:
- *rotating* i.e. increase from 0 to 255 and then wrap back to 0
- *ramping* i.e. increase from 0 to 255 then decrease back to 0, repeat

mistic100, 2022-08-25:
Yes I though about the "ramping" behavior, that's what I implemented in previous projects using directly FastLED.

But I didn't proposed it here because it would require to add tracking flags for each variable, I feel it would be too complicated. At least the "rotating" behavior is simple and already implemented, just need to be generalized.

blazoncek, 2022-10-10:
Closing as this is duplicate of #1016