wled/WLED #2768: Wi-Fi disconnection since 0.13.2

kopiro, 2022-08-29:
### What happened?

Hello!

Since 0.13.2 (or 0.13.3), my board has keeping disconnecting from Wi-Fi every 4-5 minutes when receiving UDP data from HyperHDR.

I thought that this was a problem of my board (D1 Mini), therefore I've used another one (NodeMCU) but it was giving me the same problems; 

Since some time ago the problem was not happening, I decided to downgrade to 0.13.1, and with this release the disconnection is not happening at all; it looks like this firmware is just more stable.

By pinging the device, you can clearly see this happening very 3-4 minutes.

I've also tried "Disable wifi sleep", but didn't help.

### To Reproduce Bug

Install 0.13.3, send data from HyperHDR, and wait for 2-3 minutes.

### Expected Behavior

Board should not restart.

### Install Method

Binary from WLED.me

### What version of WLED?

0.13.3

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2022-08-29:
Please attach USB to ESP and monitor serial output. You can use any utility that can read COM port (Arduino IDE or MS VSC + PIO also feature serial monitor tools).
If you do not get any output you may want to upload debug build. You can request one if you cannot find it online.

Pay attention to serial output when the disconnect happen. If you see any reset/unusual printout paste it here.

kopiro, 2022-08-29:
Ok, it actually looks like that the restart also happens with older firmware.

I'll try to give you the debug logs.

kopiro, 2022-08-29:
Alright; I've compiled the debug build and monitored... and the disconnection/reboot doesn't happen.

I've then disconnected everything and wired up to the power supply, and it happens again.

Could the problem be in the power supply itself? I'll try to monitor the voltage and check for any voltage drop.

kopiro, 2022-08-29:
Update: switching power supply with a bench power supply ( to test it ), doesn't change anything; the board start to restart, and interesting enough, when that happens, it will never recover properly.

Example of the ping:

```
64 bytes from 192.168.0.208: icmp_seq=0 ttl=128 time=85757.129 ms
Request timeout for icmp_seq 86
Request timeout for icmp_seq 87
Request timeout for icmp_seq 88
Request timeout for icmp_seq 89
64 bytes from 192.168.0.208: icmp_seq=1 ttl=128 time=89548.475 ms
Request timeout for icmp_seq 91
Request timeout for icmp_seq 92
Request timeout for icmp_seq 93
Request timeout for icmp_seq 94
Request timeout for icmp_seq 95
Request timeout for icmp_seq 96
Request timeout for icmp_seq 97
Request timeout for icmp_seq 98
Request timeout for icmp_seq 99
Request timeout for icmp_seq 100
Request timeout for icmp_seq 101
Request timeout for icmp_seq 102
Request timeout for icmp_seq 103
Request timeout for icmp_seq 104
Request timeout for icmp_seq 105
Request timeout for icmp_seq 106
Request timeout for icmp_seq 107
Request timeout for icmp_seq 108
Request timeout for icmp_seq 109
Request timeout for icmp_seq 110
64 bytes from 192.168.0.208: icmp_seq=11 ttl=128 time=101278.394 ms
64 bytes from 192.168.0.208: icmp_seq=16 ttl=128 time=97010.993 ms
64 bytes from 192.168.0.208: icmp_seq=17 ttl=128 time=96564.715 ms
64 bytes from 192.168.0.208: icmp_seq=18 ttl=128 time=97016.778 ms
Request timeout for icmp_seq 115
Request timeout for icmp_seq 116
Request timeout for icmp_seq 117
Request timeout for icmp_seq 118
64 bytes from 192.168.0.208: icmp_seq=19 ttl=128 time=100773.172 ms
```

I'll try again with the USB if it changes anything.

kopiro, 2022-08-29:
Alright; connected USB and eventually it started to happen.

But no useful messages in the logs, just a series of the same IP address (the UDP sender) that stopped at the same time the `ping` started to fail (it lost 5-6 packets), then it continued again.

I can clearly see that the board hasn't actually rebooted, is just that the Wi-Fi dropped out.

blazoncek, 2022-08-29:
Unless you get a reboot on serial (or any other meaningful output) it may be your WiFi.
UDP is known to get lost on WiFi.

kopiro, 2022-08-29:
The problem is that the WiFi of the entire board lose signal, not only UDP packets.

Phara0h, 2022-08-30:
I have the same issue, I got about 70 devices (13 wled devices various esp32 and ezp8266) on my network the only things that disconnect are WLED and it's even worse after 13.2. I am running eero mesh network and the weird thing I noticed is it says eero says still connected but I can't reach it. To get it the Wled devices to reconnect I can do 2 things. 

1. I can restart my router and that gets it to reconnect. 
2. I can restart the device that fell off the network

Notes:

* It's random what devices drop and what time they drop. I have seen as soon as 10 mins or as long as 5 hours but seems to be avg like 45min - 30min on some devices 
* Home assistant is running on my network (don't think it matters unless some erroneous packet is causing the web server to drop or wifi to stall) 
* It seems like some devices when they drop they make the other devices possibly more stable and those don't become unresponsive for longer periods of time.

blazoncek, 2022-08-30:
Mesh networks are known to cause issues.

Phara0h, 2022-08-30:
> 

Figured someone would say that even though no other ESP related device on my network has these issues, so I tried it with another wireless router thats not mesh and same issues.

balucanb, 2022-08-30:
Phara0h I was having a very similar problem using Eero and HA. I had to drop Eero as it was not playing nice and all my ESP devices just would not act right, I went back to my Google set up which is also mesh and immediately it was all Ok. HA still does not act correct all the time but I think that is a unrelated issue. My point here is mesh networks will and do work fine with WLED so unless you just have a super weak signal it is something else is my guess.

Phara0h, 2022-09-02:
> to get lost on WiFi.
I don't have any issues with my 20 other ESP devices that are not WLED. I only have issues with ESPs that are running WLED firmware.

Phara0h, 2022-09-02:
Maybe it's time to move this out of user mod and into the code base? https://github.com/Aircoookie/WLED/blob/main/usermods/Fix_unreachable_netservices_v2/usermod_Fix_unreachable_netservices.h

It's very light weight and honestly needed for most environments that use more than 10 devices on their network.

closed 2024-04-23 by github-actions[bot]