wled/WLED #2770: Everything syncing except white balance

conkydong, 2022-09-03:
### What happened?

I'm using 5 bare ESP32-WROOM-32 each with their own RGB+CCT strips and set all GPIOs to the right pins (B to 14, R to 12, G to 13, Cold White to 5, Warm White to 4). 
When I now want to sync every light everything is syncing except the white balance. If I the lights are set to cold light and I change them to warm and vice versa only the light I'm controlling is changing.

### To Reproduce Bug

See Problem.

### Expected Behavior

See Problem.

### Install Method

Self-Compiled

### What version of WLED?

every version

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Aircoookie, 2022-09-04:
Hi!
Sync should fully support white balance syncing.
Please check that the settings in the "White management" section in LED settings are the same on both units.
I don't believe it would make a difference, but you can try if enabling "Segment options" in the "Receive" section of sync settings helps.

conkydong, 2022-09-04:
I don't know what I'm doing wrong but nothing works even though the settings are the same on every light.

imtoanle, 2022-11-01:
I got this issue too. My board is ESP32 Mini (ESP32-WROOM-32).
I using PWM+CCT strip. RBG and White channel sliders worked but White balance is not.
I tried all things from sync config menu.

[image]

blazoncek, 2022-11-02:
Please post your white balance/white correction settings and sync settings.

blazoncek, 2023-09-18:
No response received, closing.

closed 2023-09-18 by blazoncek