wled/WLED #2774: ESP32 reboots (LoadProhibited)

RiverRaid, 2022-09-06:
### What happened?

Hy! I have 2 ESP32, both with the newest WLED Version v.0.13.3 "Toki" 

Both do a randomly reboot every 60 seconds to a few minutes.

### To Reproduce Bug

Apply Power

### Expected Behavior

No reboot

### Install Method

Binary from WLED.me

### What version of WLED?

 v.0.13.3 "Toki" 

### Which microcontroller/board are you seeing the problem on?

ESP32

### Relevant log/trace output

```shell
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x40158db4  PS      : 0x00060530  A0      : 0x8015979a  A1      : 0x3ffd70b0  
A2      : 0x00000000  A3      : 0x00000010  A4      : 0x000000cb  A5      : 0x3ffdd390  
A6      : 0x3ffddb51  A7      : 0x3ffddc1c  A8      : 0x801588ce  A9      : 0x3ffd7090  
A10     : 0x00000001  A11     : 0x3ffddb51  A12     : 0x000000cb  A13     : 0x00000001  
A14     : 0x00060123  A15     : 0x00000010  SAR     : 0x00000010  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000004  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0x00000000  

ELF file SHA256: 0000000000000000

Backtrace: 0x40158db1:0x3ffd70b0 0x40159797:0x3ffd7130 0x401597f7:0x3ffd7160 0x4008b506:0x3ffd7190

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
Ada
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

briodan, 2022-09-06:
are you by chance using an ESP32 module with both WiFi and ethernet? i got those messages in both 0.13.1 and 0.13.3 if i had both wifi and ethernet enabled.
I after a reset I only configured/turned one of the other its would be fine.

RiverRaid, 2022-09-06:
> are you by chance using an ESP32 module with both WiFi and ethernet? i got those messages in both 0.13.1 and 0.13.3 if i had both wifi and ethernet enabled. I after a reset I only configured/turned one of the other its would be fine.

Sorry no, just WiFI

This one are the ones i bought:
https://www.amazon.de/gp/product/B08BZGC22Q/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

softhack007, 2022-09-07:
might help if you erase the chip first, then re-install a WLED bootloader, then re-install WLED. Sometime the bootloader installed by the device manufacturers does crazy things and then crashes..

* esptool.py erase_flash
* esptool.py write_flash 0x0 ./esp32_bootloader_v4.bin
* esptool.py write_flash 0x10000 ./WLED__XXX_.bin

https://kno.wled.ge/basics/install-binary --> Flashing method 2: **esptool**

RiverRaid, 2022-09-07:
> might help if you erase the chip first, 
Hi,

of ourse i tried this already several times :)

blazoncek, 2022-09-07:
The error above usually means null pointer. As a lot of testing has been done that would possibly mean unusual set-up.
Please post your cfg.json and, if applicable, preset in question.

RiverRaid, 2022-09-07:
@blazoncek Thank you for your reply!
I have attached the cfg json as zip

Thank yoU!
[cfg.zip](https://github.com/Aircoookie/WLED/files/9508928/cfg.zip)

blazoncek, 2022-09-07:
nothing unusual.
almost same as my test set-up.

RiverRaid, 2022-09-17:
Ok, I did further testing and found the following:

As long, as the esp32 with wled is not connected to my WiFi, it works like a charm. As soon, as its connected, the reboots occur every 60 seconds.

I created a second SSID and PWD for WLED, proplem persists.

So I guess, there is something in my network that causes this... the funny thing, all esp8266 work without any Problems, it is strictly only on the esp32 (different vendors)

My Network consists of a FritzBox 7490 (not changeable, issued by my Provider) and UniFi APs.

Is there any way for me to find out what causes the Problems? I would really like to change from the esp8266 to esp32...

Thanks :)

blazoncek, 2022-09-21:
Use debug build and monitor serial output.
If you have any "smart" devices on your network try disconnecting them.
HA was known to cause issues in certain configurations.

blazoncek, 2022-11-01:
Any progress?

RiverRaid, 2022-11-07:
No, for the time no progress :( I changed to Wemos D1 and will try again in a few months

blazoncek, 2022-11-08:
Then I am going to close this issue. If the problem persists in 0.14 (when it is released) open a new issue.

closed 2022-11-08 by blazoncek

DeepflashX, 2023-04-01:
I do have the same issue. I am having two of these:
https://shop.codm.de/automation/pixel/30/wlan-pixel-controller-wled
And both have this issue. Either separate or at the same time, both have the issue (reboot every 60 seconds)

blazoncek, 2023-04-02:
Please attach crash dumps from debug build of WLED.
Please use WLED [forum](https://wled.discourse.group) or Discord for help and support questions.

DeepflashX, 2023-04-03:
And where do I find that debug build?

blazoncek, 2023-04-03:
Please use WLED [forum](https://wled.discourse.group/) or Discord for help and support questions.

codmpm, 2023-10-04:
Just referencing #2932 as the problem still persists.

@DeepflashX the problem is - as far as we know - since Fritz!OS 7.50 and only when using AVM Repeaters with mesh. Please correct me if I'm wrong.

@RiverRaid can not reproduce with Fritz 6590 and UniFi-AP's, AVM Wifi off. Neither with `0.13.3` nor `0.14b`.