wled/WLED #2775: ESP8266 reboot with Exception (29)

briodan, 2022-09-06:
### What happened?

i have an ESP8266 that reboots every 1-4 hours with an Exception (29).
it flashed with 0.13.3 from the web installer, currently does not have anything connected to it.

### To Reproduce Bug

install WLED 0.13.3 from web
connect to wifi
leave running and monitor serial port

### Expected Behavior

does not crash/reboot

### Install Method

Binary from WLED.me

### What version of WLED?

0.13.3

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Relevant log/trace output

```shell
05:25:55.050 -> --------------- CUT HERE FOR EXCEPTION DECODER ---------------
05:25:55.050 -> 
05:25:55.050 -> Exception (29):
05:25:55.050 -> epc1=0x4000df64 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000
05:25:55.050 -> 
05:25:55.050 -> >>>stack>>>
05:25:55.050 -> 
05:25:55.050 -> ctx: sys
05:25:55.050 -> sp: 3fffec10 end: 3fffffb0 offset: 0190
05:25:55.050 -> 3fffeda0:  ffffffff 3ffef390 00000002 40101328  
05:25:55.050 -> 3fffedb0:  4026d643 40105b97 3ffee090 4026d5d8  
05:25:55.050 -> 3fffedc0:  00000002 4026d57f 00000002 4026c6d8  
05:25:55.050 -> 3fffedd0:  4026c701 3fffee80 3ffef390 00000016  
05:25:55.050 -> 3fffede0:  4026a170 3fffee80 3ffef124 3ffeeaec  
05:25:55.050 -> 3fffedf0:  3ffece80 3fffee80 3fffee80 00000100  
05:25:55.112 -> 3fffee00:  656f6870 2d78696e 656d6f68 00000000  
05:25:55.112 -> 3fffee10:  00000001 00000080 3fffbdf4 00000000  
05:25:55.112 -> 3fffee20:  402664e1 00000000 00000020 d66b4f83  
05:25:55.112 -> 3fffee30:  ffffffb5 3ffeec38 3ffece90 3ffef390  
05:25:55.112 -> 3fffee40:  3ffee090 00000015 00000000 4026ae67  
05:25:55.112 -> 3fffee50:  00000000 3fff644c 3ffece80 00000000  
05:25:55.112 -> 3fffee60:  00000000 3ffef390 402727a9 00000001  
05:25:55.112 -> 3fffee70:  3fffdab0 00000000 3fffd9d0 3fff3b2c  
05:25:55.112 -> 3fffee80:  00000000 14310606 00640100 0000004b  
05:25:55.112 -> 3fffee90:  3ffecea4 000000e8 3ffecebf 3ffece98  
05:25:55.112 -> 3fffeea0:  00000000 3ffecea4 3ffeceb2 3ffecec8  
05:25:55.112 -> 3fffeeb0:  00000000 00000000 3ffecf76 3ffecf1a  
05:25:55.112 -> 3fffeec0:  3ffecedc 3ffecef8 00000000 00000000  
05:25:55.159 -> 3fffeed0:  00000000 00000000 00000015 00000000  
05:25:55.159 -> 3fffeee0:  3fff97e4 4026a8da 3ffee090 3fff644c  
05:25:55.159 -> 3fffeef0:  00000000 3ffef278 3ffee090 3ffece80  
05:25:55.159 -> 3fffef00:  3ffece80 00000110 00000000 00000015  
05:25:55.159 -> 3fffef10:  00000000 3ffece8a 40274063 3ffee090  
05:25:55.159 -> 3fffef20:  3ffece74 3fffdcc0 3ffea6e0 3ffea6e0  
05:25:55.159 -> 3fffef30:  00000080 3ffee090 00000000 3ffe888c  
05:25:55.159 -> 3fffef40:  4027391f 3fffdab0 00000000 402533cc  
05:25:55.159 -> 3fffef50:  3ffea6e0 40000f49 3fffdab0 40000f49  
05:25:55.159 -> 3fffef60:  40000e19 00000005 000b6f68 00000000  
05:25:55.159 -> 3fffef70:  3fffefc0 aa55aa55 000000ed 40105c9d  
05:25:55.206 -> 3fffef80:  40105ca3 000b6f68 00000000 dbc4870a  
[... 241 lines cut ...]
05:25:56.284 -> 3ffffea0:  00000520 000000a4 000000a4 4010175f  
05:25:56.284 -> 3ffffeb0:  3ffeff82 3fffdad0 3fff3b2c 00000000  
05:25:56.284 -> 3ffffec0:  00000000 4bc6a7f0 476c8b43 00000000  
05:25:56.331 -> 3ffffed0:  00000000 00000000 4bc6a7f0 00000000  
05:25:56.331 -> 3ffffee0:  00000000 4bc6a7f0 40101090 0036e42d  
05:25:56.331 -> 3ffffef0:  00000000 4bc6a7f0 2b851eb8 00000000  
05:25:56.331 -> 3fffff00:  3f77d45a 00000000 4bc6a7f0 00000000  
05:25:56.331 -> 3fffff10:  00000000 00000000 40101090 3fff3b2c  
05:25:56.331 -> 3fffff20:  3fff28e8 3ffeff82 3fff28d8 402133ad  
05:25:56.331 -> 3fffff30:  00000000 00000000 00000001 40100d0c  
05:25:56.331 -> 3fffff40:  3fff310c 4023ef1c 3fff28d8 3ffeff82  
05:25:56.331 -> 3fffff50:  3fffdad0 3fff3b2c 40253483 3fffefa0  
05:25:56.331 -> 3fffff60:  3fffdad0 3ffeff82 3fff28d8 40235108  
05:25:56.331 -> 3fffff70:  00000000 feefeffe 00efeffe feefeffe  
05:25:56.378 -> 3fffff80:  00000000 00000000 00000001 3fff3b2c  
05:25:56.378 -> 3fffff90:  3fffdad0 00000000 3fff3aec 40235c14  
05:25:56.378 -> 3fffffa0:  feefeffe feefeffe feefeffe 40253570  
05:25:56.378 -> <<<stack<<<
05:25:56.378 -> 
05:25:56.378 -> --------------- CUT HERE FOR EXCEPTION DECODER ---------------
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Davis8483, 2022-09-07:
I've had my instance do the same thing. It restarts maybe once or twice a day.

briodan, 2022-09-07:
a bit more details i'm seeing it on two boards with 0.13.3, I downgraded one to 0.13.1 and its has not shown any issues yet.
As a test i setup a brand new third board with 0.13.3 and will monitor

FrankRodelli, 2022-09-18:
> a bit more details i'm seeing it on two boards with 0.13.3, I downgraded one to 0.13.1 and its has not shown any issues yet.
> 
> As a test i setup a brand new third board with 0.13.3 and will monitor.
> 
> 
> 
> Edit:
> 
> I just posted this and the 0.13.1 device just crashed/rebooted with the same error.

I had 4 8266's running 0.13.3 and was also having this issue with all 4. I just flashed 0.13.1 to all 4 and I'll report back in the morning as to whether it also fixed it for me.

briodan, 2022-09-22:
update here, still saw the issue with 0.13.1

What seemed to have fixed this for me was to erase the 8266 flash with esptool then reflash with 0.13.3 or 0.13.1.
I will note that erasing flash with the webtool didn't seem to solve it, it had to be with esptool.

its been stable ever since, I'm assuming its probably related to an issue with the initial flash. These were flashed with WLED as soon as i got them so no prior binaries loaded on them.

FrankRodelli, 2022-09-22:
> update here, still saw the issue with 0.13.1
>
> What seemed to have fixed this for me was to erase the 8266 flash with esptool then reflash with 0.13.3 or 0.13.1. I will note that erasing flash with [...]

I'll give this a shot tonight. Downgrading did nothing for me, adding resistors did nothing for me, adding capacitors did nothing for me. I have a new batch of 8266's at my door for when I get home and i'll give that a shot. It's weird that I now have 5 of them that are having the same exact problem. Next step will be to hook it up to my computer to see if I find anything.

briodan, 2022-09-22:
For reference these are the ones I have, in case there is any similarities/differences to spot

KeeYees Mini Development Board for ESP8266 4M Bytes WLAN WiFi Compatible with for Arduino (5PCS) https://a.co/d/fyza9PT

FrankRodelli, 2022-09-23:
> For reference these are the ones I have, in case there is any similarities/differences to spot
> 
> 
> 
> KeeYees Mini Development Board for ESP8266 4M Bytes WLAN WiFi Compatible with for Arduino (5PCS) https://a.co/d/fyza9PT

We're not using the exact board, but I do believe I found a working setup. It's been running full brightness with effects for an hour now with no problems.

I used a 5v-3.3v level shifter and it seems to have fixed my issues. 

So the 8266 and LEDs are getting power separately from the power supply. Then I'm using the level shifter to supply 3.3v to the esp, and I have the gpio pin running through a channel on the level shifter. It seems to have worked for me and I'd be happy to share the schematics with you if you don't already have a similar setup.

briodan, 2022-09-23:
interesting if you could share schematics that would be great, not sure it will help my case since it been stable for a few days for me but might come in handy to others as well.

I was having issues with reboots with no LED's connected to the ESP8266, just ESP hanging off a basic usb power supply.
With LED's I run them of the 5V pin of the ESP8266, but the longest strip I have run on it is only 30 LED's, i'll plug in a longer one tonight and see what happens.

FrankRodelli, 2022-09-23:
> interesting if you could share schematics that would be great, not sure it will help my case since it been stable for a few days for me but might come in handy to others as well.
>
> I was having [...]

Here is a link to a video that includes the schematics: [Youtube Video](https://www.youtube.com/watch?v=ifv6wV3Rm6A)

If you need any help, feel free to reach out to me. I'd be happy to help. It does seem like running the LEDs directly from the board is not recommended, but it does work (at least for a time) My setup consists of 4 controller with; 131 lights, 300 lights, 120 lights, and 60 lights. All 4 of them would reboot within a few minutes if effects were running. The controller I followed that diagram for was running 131 and when I followed it, they have been running for 15 hours without a problem (would usually only run for 10-30 minutes before)

Davis8483, 2022-09-25:
After some testing with my setup, it only seems to reset while effects are running even if the lights aren't connected. I'm currently using gpio 4 but will try a couple of others.

blazoncek, 2022-09-25:
Use GPIO 2 if possible (or 1 oe 3).

briodan, 2022-09-25:
> Use GPIO 2 if possible (or 1 oe 3).

i'm using GPIO 4 as well any particular reason to switch of 4? ie know issues challenges with it?

blazoncek, 2022-09-25:
GPIO2 (and 1 & 3) are hardware accelerated.

Davis8483, 2022-09-25:
Just found this.

https://kno.wled.ge/features/multi-strip/#:~:text=You%20can%20use%20any%20other%20pin%2C%20but%20it%20will%20use%20the%20bitbang%20method%2C%20which%20is%20not%20recommended%20for%20reliability.%20It%20is%20best%20to%20stick%20to%20GPIO%201%2C%202%2C%20and%20if%20need%20be%2C%203.

> > Use GPIO 2 if possible (or 1 oe 3).
> 
> i'm using GPIO 4 as well any particular reason to switch of 4? ie know issues challenges with it?

blazoncek, 2022-11-18:
If you are still experiencing this issue I recommend to update to 0.14 (beta at the time) and re-open if the problem persists.
Please also follow recommendations outlined in [KB](https://kno.wled.ge).

closed 2022-11-18 by blazoncek