wled/WLED #2777: Smart custom palettes

Orace, 2022-09-09:
As discussed here [here](https://github.com/Aircoookie/WLED/issues/246#issuecomment-1241073697), custom palettes was added to [0_14](https://github.com/Aircoookie/WLED/tree/0_14).

Current custom palette JSON format is an array of integer:

    // [pos_0,red_0,green_0,blue_0,...,pos_n,red_n,green_n,blue_n]
    [0,255,0,0,...,255,0,0,255]

My proposition is to change it to an array of integer and string:

    // [[pos_0,col_0,...,[pos_n,col_n]]
    [[0,'#FFF'],[63,'#ABCDEF'],[127,'*prim'],[255,'*rnd']]

_Nested array are not mandatory._

`col_k` is a string, either:

- `#ABCDEF`: html rgb hexadecimal notation
- `#123`: html rgb hexadecimal short notation (optional)
- `*prim`,`*sec`,`*ter`,`*rnd`: reference to the primary, secondary, tertiary or random color (mnemonics to discuss).

blazoncek, 2022-09-09:
Parsing can be easily enhanced at the expense of RAM as ArduinoJson is not very efficient when handling arrays.

There is an issue with `*prim` and the others. As the primary (or other) color may change at any time, that would mean updating a RAM copy of palette each time this happens (from a prototype written in JSON). This can happen at 42 FPS if segments use different colors selected.

You could improve parsing by pre-compile JSON prototype into something faster to handle but would most likely cost additional RAM.

Do not get me wrong, I am not against such approach I am just pointing out possible issues you may encounter.

Aircoookie, 2022-09-09:
I like the idea of more color formats, yet I don't see a plausible reason for nested arrays, is a waste of memory IMO and makes it visually noisy. As you can see below, alternating index and hex string looks really clean.

For a simple red -> blue gradient, the supported color formats could be 

1) current format, which is compatible with existing gradient palettes in `palettes.h`
```

[0  , 255, 0, 0  ,
 255,   0, 0, 255]
```

2) hex or short hex notation
`[0, "#FF0000", 255, "#00f"]`

3) adding prim/sec/ter/rand is a great idea! We need to take care that color slots are visible if such a palette is active, as it is currently based on the effect which colors are available.
For mnemonics, I'd propose simply `"0"`,`"1"`,`"2"` and `"r"`

4) maybe a way of non-gradient palettes/CRGB16 support. Not a priority at all, as gradient palettes are more convenient 99% of the time

As a sidenote (this is not necessarily related to supported custom palette formats), I plan to make a `palettes.json` working similarily to `presets.json`, so it would no longer be required to have one file per custom palette, and stuff like naming custom palettes becomes possible. The only thing I'm stumped on is IDs, not sure whether to go with 255-n to 255, 100+, or with starting ID right after built-in palettes. Another option would be to have just one palette ID for custom palette and set the custom palette via a separate field. Will also add setting gradient palettes directly via API (without the necessity of saving it to FS)

Orace, 2022-09-09:
I had indeed a _palettes.json_ in mind when I wrote this proposal.
My idea is to move built-in palettes to initial value of palettes.json.
They will be modifiable/removable like any other palettes.

blazoncek, 2022-09-09:
Regarding 255-n: IMO this is optimal solution as it doesn't break anything and allows clear distinction between built-in and custom.

blazoncek, 2022-09-27:
Any progress on this?

blazoncek, 2022-11-01:
As there is no progress I am going to close both issues as it seems OP has abandoned the idea.

blazoncek, 2022-11-18:
Looks like there is no interest in pursuing this further or progress done, so I propose to close the issue.

Aircoookie, 2022-11-19:
Revisiting this, I agree that prim/sec/ter support is not trivial. No JSON parsing is necessarily required when the color changes, but the RAM copy of the palette would need to have some "magic values" that are replaced by the current colors on the fly.
Using red channel values 1,2,3 as magic values could be feasible (especially if we store custom palettes not-yet-gamma-corrected, which means all of these values would map to 0). Just a thought, a bit hacky.
See no major issue with hex string support though, will implement soon.

blazoncek, 2022-12-18:
0a0ced3e8e5c77fee0363ed8c03357a4c49aca02 may add the functionality of hex strings in custom palettes.
@Aircoookie can you review?

closed 2022-12-27 by blazoncek