wled/WLED #2778: Servo Control

digitydogs, 2022-09-13:
**Is your feature request related to a problem? Please describe.**
I have ws2812's lining the inside of all the windows in my home running WLED.  I recently decided to attempt my hand at a DIY smart blind conversion using servos, then got a bit carried away.  In order to have control of both servo and lights I either need to use ESPHome and lose all the great WLED effects, or purchase an additional esp (20 in total) for each window. 

**Describe the solution you'd like**
A setting to specify a pin as belonging to a servo motor, with fields to set a min, idle and max PWM duty cycle for the servo. 
To avoid having to map -100% through 100% to the 0-255 range, or alter the command structures, the rgb channels for a single light would translate to positional steps.  

Examples: 
-100% = 100,0,0                -50% = 50,0,0               -1% = 1,0,0
0 or idle = 0,100,0 or 0,0,0 
100% = 0,0,100                     50%  = 0,0,50                  1% = 0,0,1

**Describe alternatives you've considered**
Translating WLED effects to ESPHome Lambdas, Finding or Creating a Usermod.  Sadly Translating and creating are beyond my current confidence levels, and I haven't come across a Usermod that could fill this need.  

**Additional context**
No good effects without WLED :(
https://imgur.com/1UuePiA

Aircoookie, 2022-09-22:
Hey, that is a cool project!
As a built-in feature, servo PWM control is unfortunately outside scope of this project.
However, if someone was willing to make a usermod, it would be a great enhancement.

dosipod, 2022-10-21:
Can the PWM-Fan usermod be adapted for that ? 
https://github.com/blazoncek/WLED/issues/10

digitydogs, 2022-10-21:
It may be able to be adapted for that, unfortunately for me I am not a coder and don't know where to begin. If anyone out there is feeling kind and can tell me how to achieve the following settings via the fan usermod, or has the time to throw together a servo mod based on the fan one I would be very grateful.   

Pulling from ESPHome info would need to be able to specify the following variables

min_level : The PWM duty cycle the minimum value (-100%) will map to. Defaults to 3%.

idle_level : The PWM duty cycle the idle value (0%) will map to. This is also the state of the servo at startup. Defaults to 7.5%.

max_level : The PWM duty cycle the maximum value (100%) will map to. Defaults to 12.0%.

auto_detach_time:  The time after reaching the target value when the servo will be detached`, if set to zero, servo will not be detached. Defaults to 0s.

transition_length : The time needed for a full movement (-1.0 to 1.0). This will effectively limit the speed of the servo, the larger the value, the slowest the servo will move. Defaults to `` 0s`` This can slow down the servo to avoid loud noises or just make the movement not jerking.

Shoopity, 2022-10-30:
I'm just throwing this here because I've been looking for servo control with DMX.  Back in '16 I had a very basic proof of concept to use Vixen (audio/lighting synch show program, mainly for Christmas shows) to control servos.  I think [this is the code](https://gist.github.com/hsiboy/7e23ec783d8b2131133f) I found, and, after adding in the ability control servos (calling #include <Servo.h>, defining the pin, all the normal things you have to do to drive a servo with an arduino), I just changed the data that came from the DMX (which is a range from 0~255) into something that's 0~180, then install of calling FastLED.Show(), I called myservo.write(...).

I don't _think_ it would be that hard, and I totally understand it's outside the scope for this to be something the official coders include, but I just need to be pointed in the right direction.  Is the DMX data already ingested and in my user mod I just need to massage it, or does my user mod need to ingest it... again?  Where is the equivalent of FastLED.Show() called in WLED?  Can I just do a myservo.write(_blah_) in my user mod?  There's currently 14 different types of outputs to choose from... how I define a new type of output, in this case a servo?

digitydogs, 2022-10-30:
> t I just need to be pointed in the right direction.

If you manage to get this sorted would you be so kind as to share your usermod? I have got to get off of ESPHome as its E131 implementation is both lacking and laggy.

BigBill2000, 2022-12-17:
Hi - I thought I would comment as I have figured out a way of getting a servo working. This may not be the best way as I am not that good in coding, but I am learning! In current setup there is a "PWM White" output, I am using that to trick it into working. I then modified the code in two files, const.h and bus_manager.h. Basically I change the PWM White frequency to 50hz, then change the output from 0-255 to 5-30 to get the pulse width 1ms-2ms range as servo requires. You may need to adjust the 5-30 per your specific servo.

If I knew how to do a usermod, I would create something other than "PWM White" and make it "PWM Servo" and make the 50hz and 5-30 both an adjustable range, but I don't know how. Maybe will work on figuring out how to do that, but that is beyond my level at this point.

const.h
// PWM settings
#ifndef WLED_PWM_FREQ
#ifdef ESP8266
  #define WLED_PWM_FREQ    880 //PWM frequency proven as good for LEDs
#else
  **#define WLED_PWM_FREQ  50 // Bill default 19531**
#endif
#endif

bus_manager.h
  void show() {
    if (!_valid) return;
    uint8_t numPins = NUM_PWM_PINS(_type);
    for (uint8_t i = 0; i < numPins; i++) {
      uint8_t scaled = (_data[i] * _bri) / 255;
      if (reversed) scaled = 255 - scaled;
      **uint32_t bscaled = map(scaled,0, 255, 5, 30); //Bill 5-30 is full range**
      #ifdef ESP8266
      analogWrite(_pins[i], scaled);
      #else
      **ledcWrite(_ledcStart + i, bscaled); //Bill**
      #endif
    }

blazoncek, 2022-12-17:
Please write a usermod.
You may take PWM fan, boblight, Multi relay and some others as a reference on how to do things,

dosipod, 2023-04-19:
https://github.com/Aircoookie/WLED/tree/main/usermods/pwm_outputs

softhack007, 2023-10-07:
Closing as the pwm_outputs usermod is (partly) supporting the request. No plans to integrate PWM motor / fan control in the WLED core,  besides the existing usermod.

closed 2023-10-07 by softhack007

EColi01, 2025-08-08:
Will there be support for esp8266 for the user pwm outputs?

DedeHai, 2025-08-08:
unlikely. ESP8266 is end of life, just get an ESP32.