wled/WLED #2784: It seems not working stably with Openwrt

Doublefire-Chen, 2022-09-16:
### What happened?

I am using my Openwrt soft router as a DHCP server, but sometimes I can't open the management web with the ip address. Ping is success, nmap can't scan any port open on my ESP32. But when I am using a normal router with DHCP model, everything is working well.

### To Reproduce Bug

Let EDP32 connect to a router which is operating on  Openwrt system, then can't open the management. This problem is not all the time, sometimes you can open the page. So please let me describe it as not stable.

### Expected Behavior

Management page is not working.

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.13.3(Built number I don't know as I can't open the page right now)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Anything else?

Board: ESP32 Devkit v1
[image]
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Doublefire-Chen, 2022-09-16:
Just moment I installed the system with bin file, same problem.

Doublefire-Chen, 2022-09-16:
Just moment I installed the WLED 0.13.1 system with bin file, same problem.

Doublefire-Chen, 2022-09-16:
I remember last time I solve this problem by restarting the router.

Doublefire-Chen, 2022-09-16:
> I remember last time I solve this problem by restarting the router.

Just moment I solved this by restarting the router, but I am afraid when this problem happen again.

Aircoookie, 2022-09-16:
Thank you for the issue report. Just to confirm that the connection issue is indeed caused by DHCP, have you tried if the issue occurs after setting a static IP within WLED settings ?

Doublefire-Chen, 2022-09-18:
> Thank you for the issue report. Just to confirm that the connection issue is indeed caused by DHCP, have you tried if the issue occurs after setting a static IP within WLED settings ?

This issue occurs randomly, and now I am using it well. I remember once I tried with static ip connecting my soft router when this issue happened, but did't work.

Doublefire-Chen, 2022-09-19:
It is a quite annoying thing.
Now it happened again and I noticed that the blue led on my ESP32 is twinkling all the time right now.

Doublefire-Chen, 2022-09-19:
> It is a quite annoying thing. Now it happened again and I noticed that the blue led on my ESP32 is twinkling all the time right now.

I just solve it by restarting my router and ESP32, and I find that it is not matter with the blue led the board as it still twinkle though it is working well. I don't know when is next time.😭 Maybe I should install a new version of Openwrt to prove it is the problem of soft router's system.

blazoncek, 2022-09-19:
Just get a good router from Ubiqiti, MikroTik or the like.
You can get EdgeRouter X for less than $50.

mahipat99, 2022-09-20:
I have the same issue. ESP8266 goes offline randomly
Tried resetting/upgrading router FW multiple times, disabled blynk, hue, etc. , changed power supply / ESP 
nothing helps

As soon as I restart wifi radio, the device comes online but then goes offline randomly.

The only workaround I found is static IP from WLED itself, I have tried setting static IP from the router, too, but it also didn't work
one more thing, I have two routers, one is router+AP, and another is dumb AP(disabled DHCP). Workaround works only on router+AP; same disconnect issue with dumb AP.

FW - 22.03.0 (Mi Router 4A)

Jason2866, 2022-09-25:
It is not a general problem. I just can say we do not have this issue with Tasmota.
No reconnects or any other stability issues. I have OpenWRT 22.03 and 19.xx running on 2 AP (Xiaomi 4a gigabit) and in the setup 4 other APs (business).

closed 2022-09-26 by Doublefire-Chen

Doublefire-Chen, 2022-09-26:
Problem solved by turning off firewall on my solid router.