wled/WLED #2786: Can this be adapted for LED strip shapes?

timof123, 2022-09-17:
I want to use this firmware for homemade snowflakes from WS2812B LED strip. The fact is that many effects (for example, "Chase") do not look quite correct due to the peculiarities of connecting the tape. Is it possible to change the default LED sequence in the firmware? That is, I need, for example, LED #1 in the firmware to be equal to the physical LED #12; #2 equaled #13, and so on. Is it really possible to do this?

blazoncek, 2022-09-17:
Please use [forum](https://wled.discourse.group) or Discord for help and support questions.
Closing

closed 2022-09-17 by blazoncek

Aircoookie, 2022-09-17:
https://kno.wled.ge/advanced/mapping/

sansillusion, 2022-09-18:
Look into ledmap
I never used it but saw it was implemented a while ago.