wled/WLED #2790: usermod for Sleep?

eried, 2022-09-20:
Hi! I need to make the esp8266 or ESP32 fully sleep so I am just going to trigger sleep after some time. However:

any way to see if there is any *actual* activity in wled?

eried, 2022-09-20:
I know, I will press reset. Activity would be, user clicking on the WebUI or some command via network. If is just playing an effect I would consider this as idle (and start adding this for the inactivity timer)

blazoncek, 2022-09-21:
IMO the whole purpose of WLED is displaying effects (AKA lighting up LEDs). I would consider *this* activity and not user interaction.
If all you care is UI interaction you get notification via `readFromJsonState()`. Each time a JSON API call happens usermod is notified by call to this function.

eried, 2022-09-21:
Cool, thanks :)

closed 2022-09-21 by eried