wled/WLED #2792: Random pallet cycle not working

TysTies, 2022-09-21:
### What happened?

 I can't get the random pallet effect to work on any effects. I wanted to use the Dissolve effect with a random pallet cycle, but it won't work. It just stays on the default one.

### To Reproduce Bug

Open app or whatever then use dissolve effect then click random pallet it just stays on the same one. 

### Expected Behavior

I thought it was a cycle.... of random pallets 

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.13.3 (build2208222)

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2022-09-21:
This is solved in forthcoming 0.14.

closed 2022-09-21 by blazoncek

TysTies, 2022-09-21:
When will that be released?