wled/WLED #2795: ESP-01S never connects to wifi

Renohedron, 2022-09-23:
I tried to make it run for days now but no matter what, it never connects to my wifi and I found no way how to check logs or somehow find what's the issue for WLED.
1) I flashed WLED_0.13.3_ESP01.bin many times both with web and esptool-v3.3.2, always erasing the flash before
2) I created _exactly_ same network with some ancient **no-name AP and it connects well**
3) My AP is years old Vodafone-branded cable modem/AP/router box made by Compal
4) There are no special characters, channels or features (I know of) enabled on the Compal
5) While my DHCP server knows the mac address and offers the IP address, I cannot ping or reach it via browser, there is simply no response from the device

Any ideas, how to debug that connection issue, please?

softhack007, 2022-09-23:
Maybe you already checked this - ESP only uses 2.4GHz band. So in case you wifi router has 5GHz enabled but no 2.4GHz, then ESP will never find the router.

Renohedron, 2022-09-23:
Thanks, that's what I tried to cover in 2). Both networks do not only have the same name and password, but also frequency, channel, channel width, mode...

xbmcnut, 2022-10-06:
I too had similar issues with 0.13.3 on ESP-01M. Could connect to my 2.4Ghz WiFi but upon reboot, it looses the WiFi settings. On top of that, backup config did not work nor did presets. Rolled back to 0.13.2 and everything is working again. I fully erased the board before installing 0.13.3 initially with the web installer and then a 2nd time with ESP-Flasher. Both methods resulted in the same failed operation.

As a side, the presets configured in 0.13.2 don't seem to be getting pushed through to the Home Assistant integration as the presets presented in HA do not match what's configured in WLED.

Renohedron, 2022-10-08:
I'm pretty sure I tried older versions, but it's been a while, I can't remember exactly. I'll try 13.2.

specs32, 2022-10-10:
can you try another esp module ? .. I ordered 5 esp-01 modules and one showed the same symptoms as described here .. it tried to connect but failed over and over... the next module out of the pack ran just fine...

xbmcnut, 2022-10-11:
I have confirmed there definitely seems to be an issue with 13.3 on ESP-01M. I purchased another board to test and like before, WiFi does not stick on reboot. Rolling back to 13.2 fixes the issue.

Renohedron, 2022-10-11:
I have older, plain ESP-01 which can't be flashed for same reason (fails around 93% each time), but otherwise, no other unit to try.

At least I tried 13.2 like this:

<pre><code>
<b>PS C:\Users\renmc\OneDrive\Software\esptool-v3.3.2-win64></b> .\esptool.exe -p COM3 read_mac
esptool.py v3.3.2
Serial port COM3
Connecting....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting...
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 94:b9:7e:13:e8:ff
Uploading stub...
Running stub...
Stub running...
MAC: 94:b9:7e:13:e8:ff
Hard resetting via RTS pin...
<b>PS C:\Users\renmc\OneDrive\Software\esptool-v3.3.2-win64></b> .\esptool.exe -p COM3 erase_flash
esptool.py v3.3.2
Serial port COM3
Connecting...
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting...
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 94:b9:7e:13:e8:ff
Stub is already running. No upload is necessary.
Erasing flash (this may take a while)...
Chip erase completed successfully in 2.4s
Hard resetting via RTS pin...
<b>PS C:\Users\renmc\OneDrive\Software\esptool-v3.3.2-win64></b> .\esptool.exe -p COM3 write_flash 0x0 .\WLED_0.13.2_ESP01.bin
esptool.py v3.3.2
Serial port COM3
Connecting...
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting...
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 94:b9:7e:13:e8:ff
Stub is already running. No upload is necessary.
Configuring flash size...
Flash will be erased from 0x00000000 to 0x000b4fff...
Compressed 739296 bytes to 499867...
Wrote 739296 bytes (499867 compressed) at 0x00000000 in 44.2 seconds (effective 133.8 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
</code></pre>

Configured via AP mode from my mobile phone but no luck, after confirming restart, the device is gone as always, so 13.2 behaves same as 13.3 here. I even check dhcp traffic this time and something definitely happens:

```
Oct 11 21:51:11 calcium dnsmasq-dhcp[35692]: DHCPDISCOVER(enp0s31f6) 94:b9:7e:13:e8:ff
Oct 11 21:51:11 calcium dnsmasq-dhcp[35692]: DHCPOFFER(enp0s31f6) 192.168.0.40 94:b9:7e:13:e8:ff
Oct 11 21:51:31 calcium dnsmasq-dhcp[35692]: DHCPDISCOVER(enp0s31f6) 94:b9:7e:13:e8:ff
Oct 11 21:51:31 calcium dnsmasq-dhcp[35692]: DHCPOFFER(enp0s31f6) 192.168.0.40 94:b9:7e:13:e8:ff
Oct 11 21:51:47 calcium dnsmasq-dhcp[35692]: DHCPDISCOVER(enp0s31f6) 94:b9:7e:13:e8:ff
Oct 11 21:51:47 calcium dnsmasq-dhcp[35692]: DHCPOFFER(enp0s31f6) 192.168.0.40 94:b9:7e:13:e8:ff
Oct 11 21:52:29 calcium dnsmasq-dhcp[35692]: DHCPDISCOVER(enp0s31f6) 94:b9:7e:13:e8:ff
Oct 11 21:52:29 calcium dnsmasq-dhcp[35692]: DHCPOFFER(enp0s31f6) 192.168.0.40 94:b9:7e:13:e8:ff
Oct 11 21:54:07 calcium dnsmasq-dhcp[35692]: DHCPDISCOVER(enp0s31f6) 94:b9:7e:13:e8:ff
Oct 11 21:54:07 calcium dnsmasq-dhcp[35692]: DHCPOFFER(enp0s31f6) 192.168.0.40 94:b9:7e:13:e8:ff
Oct 11 21:54:25 calcium dnsmasq-dhcp[35692]: DHCPDISCOVER(enp0s31f6) 94:b9:7e:13:e8:ff
Oct 11 21:54:25 calcium dnsmasq-dhcp[35692]: DHCPOFFER(enp0s31f6) 192.168.0.40 94:b9:7e:13:e8:ff
```

So WLED is present in the network, but handshake probably doesn't work as it should.

blazoncek, 2022-11-01:
ESP01 (S or non-S) are dirt cheap and mass produced. As someone who owns about 100 of them I can tell you that manufacturing quality varies substantially.

Some you will not be able to flash at all, some will not boot after flashing and then some will start WLED but never connect to WiFi.

If you find that particular ESP01 only works with particular WLED version this may be due to te Arduino ESP core version used with WLED. As we move forward there is no way we can support any odd HW and use different Arduino cores. But you can try to compile yourself changing options in platformio.ini (or override).

Closing.

Renohedron, 2022-11-06:
Heard and understood. I just want to confirm for everybody, new batch of ESPs arrived a few days ago and the very first one immediately works as expected, so both original units were somehow faulty.
Obviously, if this is your first experiment in the new world of IoT and controllers and so, it's easy to get discouraged if you know you repeated everything 20 times properly and still didn't succeed. I am proof that @blazoncek speaks the truth, the quality of those cheap generic boards varies a lot. Just buy more from different sources to avoid difficulties I experienced and good luck with your endeavors.

blazoncek, 2022-11-06:
Out of 10 that I tried to install into corner lamps that I make, 4 were faulty.

TheAndi, 2022-11-17:
### Possible Solution for people who stumble across that and want to get it working on an ESP-01S
(at least worked for me with an ESP-01S that can run 0.13.2)

- Download Source of new version and change Arduino ESP core version (thanks @blazoncek for suggesting) to use for ESP8266 in `platformio.ini`
from `platform_wled_default = ${common.arduino_core_3_2_0}`
to       `platform_wled_default = ${common.arduino_core_2_7_4}`

- If you get `PIO Core Call Error` (like me) try restarting VSCode/PIO (or see [#2123](https://github.com/platformio/platformio-home/issues/2123))
- If then fails to find the device at correct COM Port (like it happened for some reason) it's always possible to flash the compiled binary from `.pio\build\esp01_1m_full\firmware.bin` using ESP Tool as explained [HERE](https://kno.wled.ge/basics/install-binary/)

Hope this might help someone. At least my ESP-01S now runs the newest WLED release 0.14 without issues :)

TheAndi, 2022-11-17:
True, I also tested all the flash modes because I found this while researching. But no flash mode solved my issue, but good hint, I'll update my previous post. Maybe this Info can then be linked to the official docs somehow, to help users still using ESP-01S boards.

Renohedron, 2022-11-17:
Hm, as a total newbie I barely understand a word in those last posts. I'll try to educate myself a bit but first, is ESP-01S something outdated..? Was it somehow replaced or...?

TheAndi, 2022-11-18:
I think once you figure out to compile and flash it yourself, the steps I listed should be quite easy to do.
Maybe this gets you started: https://www.youtube.com/watch?v=fRxyp5Yhivc