wled/WLED #2796: LEDs go crazy after 50

Buschmxnn, 2022-09-25:
### What happened?

Hey!
So I'm using WLED for over a year now and so far no problems.
But since this week my bar goes crazy after like the 50. LED.
I reflashed my ESP8266 bc I thought maybe some bug has caught it, but no sign of relieve.
I am using 600 LEDs, the setup is the same as before, so there shouldn't be the issue. 
Funny enough, when I set the bar on pure red, I am having no issues, no flickering, nothing. But any otner colour or effect results in weird flashing. I tried to play with the setting such as setting RGB to any other, but nothing helped. I have a RGBW LED strip!

Thanks for helping! :)

### To Reproduce Bug

Any mode except solid red. 

### Expected Behavior

The bar should show the modes in the meant way, not result in weird light flashes.

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.13.3 ESP8266

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2022-09-25:
Did you try another LED strip? Does it behave the same?
If not, you may have a faulty LEDs somewhere on the strip.

Buschmxnn, 2022-09-26:
Weirdly enough, when i pulled the strip out of the bracket, it worked just fine. I'm pretty sure the strip has a problem of some kind. Hope it stays working now, otherwise i gotta trim it and solder it together. Thanks anyways! :)

closed 2022-09-26 by Buschmxnn

blazoncek, 2022-09-26:
Perhaps a bad solder joint on a single LED. Twisting or bending the strip may have temporarily solved the issue but it may re-appear at any time.