wled/WLED #2798: DDP output hostname instead of IP

roboraptor, 2022-09-27:
**Is your feature request related to a problem? Please describe.**
Having Wleds on network with changing IPs.

**Describe the solution you'd like**
Possibility to set hostname instead of IP address, when setting up DDP output for different WLED nodes.

**Describe alternatives you've considered**
Setting up static IPs but that would not work when carrying the lights around.

Aircoookie, 2022-09-27:
Resolving a dynamic IP from the hostname is probably not feasible, but doing so from the mDNS address should work. 
However, this requires storing the mDNS name string in the network bus.

roboraptor, 2022-09-27:
So, it would be possible?
It would be very helpful when moving lights around every second week.

blazoncek, 2022-09-27:
Possible, yes, feasible, no.

roboraptor, 2022-09-27:
Not even the mDNS?

closed 2022-11-01 by blazoncek