wled/WLED #2805: Setting analogue clock colours - Idea

smartroad, 2022-10-01:
I use the analogue clock overlay and have changed the colours of the "hands" by editing the source hard coded values. I was thinking would it be possible to add a box in the segments section called "Analogue Clock" when it is switch on in the settings? My idea is that it would be something like this:

[image]

Where you can select the overall brightness (if you wanted it dimmer at night for example), set the colours for all the hands and markers and choose to have the markers on or off. These could then be saved when setting as a preset can called back at any time.

Clicking on the coloured boxes could bring up the colour wheel or maybe just copy over the current selected colour on the colour wheel?

dikkedimi, 2023-12-26:
Would love this, even if it was just an "invert colors" option, it's the only thing I'm missing in WLED compared to the velleman brightdot clock firmware.  additionally the option to choose any existing color scheme would be nice. I actually like cyan/magenta/yellow more than I like the primary RGB scheme 😊 .

It seems crazy to me this feature isn't implemented already, but I guess not a lot of folks use WLED to tell the time (I know from experience you lose at leaste a minute figuring out what time it is 😅 )

jasmohan-narula, 2024-07-04:
I also want this.

blazoncek, 2024-07-05:
There is a usermod for analog clock overlay which is more flexible and allows customization (#2736).
Built-in analog clock overlay may be removed in future updates (due to memory constraints).

closed 2024-07-05 by blazoncek