wled/WLED #2806: Wled 0.13.3 crashes after update from 0.13.2

ferrarid61, 2022-10-02:
### What happened?

I upgrade my esp8622 d1 mini from 0.13.2 to 0.13.3. Now wled crashes after few seconds.
I did it with home assistant

### To Reproduce Bug

Upgrade from 0.13.2 to 0.13.3

### Expected Behavior

Working fine

### Install Method

Binary from WLED.me

### What version of WLED?

0.13.3

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2022-10-02:
Please, download binary from GH and update manually.

ferrarid61, 2022-10-02:
I did a manual upgrade to 0.13.3 but still crashes after 20 seconds. Any ideas? How problem on my esp8622? Is there any log to look at?

blazoncek, 2022-10-03:
I know it may be destructive, but you can backup presets and configuration from /edit page, so I would recommend erasing flash and doing a clean install.
You can restore presets and configuration afterwards.

ferrarid61, 2022-10-03:
I did a lot of troubleshooting and I think to have found the crash scenario. Il happens when I use the webapp on IoS on my Ipad. It works perfectly on PC and Android webapp
I did a clean installation erasing esp micro and installing a fresh 0.13.3

ferrarid61, 2022-10-04:
I confirm it is a problem on webapp on IoS. Closing the issue

closed 2022-10-04 by ferrarid61