wled/WLED #2807: Crazy Flickering with WLED on WT32-eth01 board (esp32 with ethernet)

PixelLights, 2022-10-02:
### What happened?

I'm encountering a serious flicker issue on 3 outputs using WLED on the WT32-eth01 board.

This is not a physical problem (the distance from the led strip - signal wire - to the board is less than 20cm, I am using the SN74HCT245N as a level shifter and all the negatives are connected together). All other outputs work fine with this same configuration. I tested it on 3 different WT32 boards and they had the same problem

The problem happens on outputs 6 to 8, but if you use fewer outputs it happens in a variety of ways too. It's simple to visualize the problem recriating my configuration, but using fewer outputs in other ways, all of leds flickers in random times, so it's not a problem with the pins on WT32.

The best way to visualize the problem is to make the same settings as in the post and use a program on the computer to light only one of the problematic outputs (6, 7 or 8)

The same problem occurs in some automatic WLED effects, mainly the "police" effect. This excludes the possibility that there is a problem with the reception of the E1.31 signal.

I did the tests with the WS2815 12V led strip and also the 5V WS2812B. Both had the same problem.

If I remove the 2 last outputs (9 and 10) the flickering on channels 6 - 8 stops, but sometimes all of the leds flickers randonly. Using just 6 outputs with more leds per output, creates some flicker on all of the leds too (in random times). The best way to visualize the problem is to recreate my configuration and light up just one of the flickering outputs (6 to 8).

I've tried enable and disable all of the configurations from all of WLED menus, but none seems to reduce the problem.

The only crazy solution that worked was to use a computer program to send the E131 signal, turn on all the LEDs with the minimum brightness in white color, that is, from 1 to 255, send the signal at level 1 to red, 1 for green and 1 for blue and then create the effects normally, at maximum brightness, "on top" of these lit LEDs, that is, adding up. So flickering ends up in this 10-channel configuration. In short, in effects where all the LEDs are lit, there is no flicker, but in effects where only a few LEDs are lit, flicker occurs.

Just to reinforce, this happens in WLED auto effects too, with the ethernet cable disconnected, so it's not a problem with the program that sends the E131 signal; It's just easier to visualize the problem using an external program.

My configuration:

108 pixels per output, RGB led order

Output order from 1 to 10:
1: PIN 2
2: PIN 4
3: PIN 12
4: PIN 14
5: PIN 15
6: PIN 17
7: PIN 5
8: PIN 33
9: PIN 32
10: PIN 3

### To Reproduce Bug

Use the same configuration

### Expected Behavior

No flicker

### Install Method

Binary from WLED.me

### What version of WLED?

Tried on WLED 0.13.3 and WLED 0.13.2

### Which microcontroller/board are you seeing the problem on?

Other

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Aircoookie, 2022-10-22:
Thank you for the report!
I have no immediate solution unfortunately, but I am inclined to believe this issue is caused by https://github.com/espressif/arduino-esp32/issues/2885

softhack007, 2022-10-27:
related to #2542 ?

softhack007, 2022-11-26:
If I understand the specs correctly, GPIO 17, 5, 32, 33 all have special purposes for controlling the on-board ethernet chip RX2, TX2, CFG, 485_EN.

It might be these PINs cannot be used for connecting LEDs, as you can't cut their on-board connection to the ethernet chip.

https://github.com/espressif/arduino-esp32/blob/fd72cf46ad6fc1a6de99c1d83ba8eba17d80a4ee/variants/wt32-eth01/pins_arduino.h#L34-L40

closed 2024-04-23 by github-actions[bot]

PixelLights, 2024-10-11:
Has anyone achieved a definitive solution to this problem?

softhack007, 2024-10-11:
Nothing I know about, sorry.

The problem report report was closed by the bot , because there was no activity (or expressed interest by original author) for a long time.

foddy-foo, 2025-09-14:
I have seen the same behaviour when using WT32-ETH01 with WS2812b strips. I have two such boards and I tried using them with WLED but got strange flickering. I have made a lot of WLED installations without such issues when using non WT32-ETH01. I always use a short data line and logic level shifters. In the end I gave up on the ETH01 units.