wled/WLED #2811: Random color and effect jumps in ArtNet mode

martinkoenig, 2022-10-03:
### What happened?

I´m using WLED over ArtNet with DmxControl 3 in Effect Mode. 
Color palettes and effects jump randomly for a brief second without touching any DMX values. It´s like a flash. 

Let´s say I´m using effect "Wipe Random" with color palette "Retro Clown". Then, from time to time, it flashes to color palette "Yelblue" and back. You can watch this behavior in the WebUI.

This makes WLED as an ArtNet fixture unusable

### To Reproduce Bug

- Flash prebuild v0.13.3 binary or self compiled "0_14" branch (happens for both)
- Go to sync settings and switch DMX Type E1.31 to ArtNet and DMX Mode to Effect
- Setup WLED Device as ArtNet Interface in DmxControl 3
- Select some effect and color palette

### Expected Behavior

Well, no flashes between color palettes. It should stay at the set effect and color palette without flashing to some other colors and back.

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.13.3 (build 2208222)

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2022-10-03:
A bit more to the right, in `"seg":{...` you should see "pal" entry. If it is changing it means something is changing palette.
WLED does not do it on it's own.

martinkoenig, 2022-10-03:
Indeed, pal entry is changing. So DmxControl is the one to blame... I will investigate this!
Sorry for wasting your time!

And btw, I frickin love WLED, you are doing an awesome job!

closed 2022-10-03 by martinkoenig

blazoncek, 2022-10-03:
Try firing up WireShark and sniff network packets.