wled/WLED #2813: Support for WB2S / Beken BK7231 (or LMA35?) Chip

pegelf, 2022-10-04:
Hi. I've bought an 8€ noname 10m rgb fairylight on [Amazon](https://www.amazon.de/gp/product/B08LT424F3/), which has 3 wires and is WLED-capable. You only have to cut off the controller and connect the 3 wires to an ESP instead.

The original controller also has Wifi and and an IR-receiver, so I asked myself if I could also flash WLED on that instead. This would be great, because that would make the fairylights including a controller very cheap.

So I took it apart and saw this:
[image]
[image]

I removed the metal plate on the chip, to see what is written there:
[image]
[image]

It says "Beken BK7231TQN32 AU035AFB". There also seems to be an open source project for it:
https://github.com/btsimonh/tuya-iotos-embeded-sdk-wifi-ble-bk7231t

Are there any plans to support that chip? I also don't know how much work that would take, I'm not able to program that and I know that open source projects are a hobby, so of course I can't demand that WLED will ever support that, but I thought I could at least create that issue, so people know that there are commercial products with this chip inside.

blazoncek, 2022-10-04:
Most likely no.

Aircoookie, 2022-10-04:
Would be possible if there was a stable Arduino core and async network libs for it, but yeah, right now BK7231T support seems very unlikely.

closed 2022-11-01 by blazoncek

spupetic, 2024-02-03:
As OpenBK7231T_App and LibreTiny has come a long way since this was closed, and the newer Tuya controllers are usually based on the Beken 7231N or the 7231T, it would be nice to support them.

broyuken, 2024-05-16:
If I do a replacement where I remove the BK chip and pop in an esp8266, would this work? Or would I need to remove or bypass the Tuya MCU?