wled/WLED #2815: Effects not working with JSON API

cylon147, 2022-10-06:
### What happened?

I am trying to set the effects using JSON API but somehow the Effects are not changing I am getting the success Replay from Server but it's not executing the Effects.

The Command I am Sending
`{"effects":["Flow"]}`

Response from Server
`{"success":true}`

### To Reproduce Bug

Send JSON API Command
`{"effects":["Flow"]}`

### Expected Behavior

The Effects should change.

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.13.3 (2208222)

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2022-10-06:
Please use official JSON API available at https://kno.wled.ge/interfaces/json-api/
Closing.

closed 2022-10-06 by blazoncek

cylon147, 2022-10-08:
What is the API syntax for setting Effects via JSON API? The Documentation is not clear about this. Please Help.

blazoncek, 2022-10-08:
Serialized JSON. No special tricks there.

cylon147, 2022-10-08:
I have tried sending `{"effects":["Blink"]}` both as JSON and Text but none is working.
I do get `{"success":true}` response from the server but the effect is not getting Executed.

[image]

[image]

blazoncek, 2022-10-08:
As said above: Use official API as described in KB.
Also, use your browser's developer tools to observe WLED UI submitted commands.
Please use WLED [forum](https://wled.discourse.group) or Discord for help and support questions.

sn1ce, 2022-10-28:
Hi there,

i am also struggling, since its not working for me neither.
A simple " you need to do it like this" would help, since not all of us are familiar with json and stuff.

In my case, i wanted to use toggle next effect, i am able to do it with HTTP API (easy one##   /win&A=160&FX=~   ), but not able to make it with the json API.

Helping out here, much appreciated. 

sn1ce

sn1ce, 2022-10-28:
@cylon147 
Hi again,

I was able to set it up to "Flow" with  the following, {"seg":[{"fx":"110"}]}
Make sure, you check on "your.ip.of.wled/json/eff ; there is a newline after the 10th entry, so you can count to your effect you want to have :-) 

Cheers